Atmosphere-NX/Atmosphere #777: Fatal Error

alexqd3, 2019-12-25:
Im sorry to be such a newbie but im really lost and do not know what to do.

My Switch is stuck on a black screen with the fatal error message:

Switch Firmware 9.10

A fatal error occurred when running Atmosphere.
Title ID: 010041544d530000
Error Desc: std: :abort() called  (0xcffe)
Report saved to /atmosphere/fatal_errors/report_000000000656188e.bin

Press Power to reboot.

This happened when trying to update my Atmosphere to the latest version.
I really dont know where to start and could use some help, please.

ObscenityIB, 2019-12-25:
Many people have been having this issue lately, it seems to spring up from nowhere and for no reason.

Things suggested so far have been to make sure both Atmosphere and Hekate are the latest.
Then fix BOTH archive bits in Hekate or manually in your OS.
Then if that doesn't work, try making sure `sdmc:/atmosphere/contents/` is empty, then put the stock contents from the last update in there again

Best of luck to you.

alexqd3, 2019-12-26:
Thank you for responding to me

> Many people have been having this issue lately, it seems to spring up from nowhere and for no reason.
>
> Things suggested so far have been to make sure both Atmosphere and Hekate are the latest.
> Then [...]

Thanks for the reply. My Switch was modded with a chip. Not sure what Hakate is. This there a guide that shows you what you need from start to end? All of this is very new to me.

ObscenityIB, 2019-12-27:
> My Switch was modded with a chip

wait whaaat? thats a thing? i never knew any hard mods existed.
everyone i know just softmods, because its free and easy.

>This there a guide that shows you what you need from start to end?

for you, i have no idea. for everyone else, [its this.](https://nh-server.github.io/switch-guide/)

SciresM, 2019-12-30:
Report binary wasn't uploaded, closing due to lack of info.

closed 2019-12-30 by SciresM