Atmosphere-NX/Atmosphere #779: unknown pg1 version. failed to launch HOS

thomvb21, 2019-12-27:
## Bug Report

tried to jailbreak my switch on firmware 9.1.0 with hekate 5.1.1 
i tried to setup emummc and when i try to launch it i get the error message:

unknown pkg1 version.
failed to launch HOS!
press any key . . .

and then it launched me back to hekate

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - 9.1.0", ]</br>

### What's the issue you encountered?

[  am getting error when loading into emmummc 

Unknown pkg1 version.
Failed to launch HOS!
Press any key . . .

Please help me 

### How can the issue be reproduced?

launching into emummc.

Please help me## Crash Report

 am getting error when loading into emmummc 

Unknown pkg1 version.
Failed to launch HOS!
Press any key . . .

Please help me

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

HookedBehemoth, 2019-12-27:
Your Atmosphère version is out of date.

thomvb21, 2019-12-27:
> Your Atmosphère version is out of date.

i downloaded the newest atmosphere version from the website and put it on my sd card

fincs, 2019-12-27:
> from the website

The only official Atmosphère download location is https://github.com/Atmosphere-NX/Atmosphere/releases/latest

SciresM, 2019-12-27:
That's a hekate error.

closed 2019-12-27 by SciresM