Atmosphere-NX/Atmosphere #780: Can't access Applet Mode

PKMWM1, 2019-12-27:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>
I am on the latest version of atmosphere and switch firmware. Although I can access homebrew through the standard title takeover method, I can't access homebrew through applet mode at all. Whenever I hold the R button and select album, it just shows me my pictures. I tried rebooting and re injecting the payload and it still isn't working. I can't find any notes about this online. Any help would be appreciated.
[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

HookedBehemoth, 2019-12-27:
From the release page:

- Default override behavior was changed.
  - By default, atmosphere will now override the album applet with hbl unless R is held.
  - By default, atmosphere will now override any application with hbl only if R is held.

closed 2019-12-28 by SciresM

simonrb2000, 2020-01-01:
I'm on latest and mine still behaves the old way for album. As in I have to hold R to get Applet Mode. I can't run any title on tack over mode though.. Strange