Atmosphere-NX/Atmosphere #781: Abort when invoking any web applet inside applet mode

amicuchu, 2019-12-27:
## Bug Report

### What's the issue you encountered?

Opening a web applet inside applet mode ends in fatal error

### How can the issue be reproduced?

1. Enter in hbloader using the album so it's under applet mode
2. Execute Browser.nro
3. Wait for program to invoke web applet

### Crash Report

Fatal error generated 
[report_15a12411993.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4006419/report_15a12411993.zip)

### System Firmware Version

System version: 9.0.0
Atmosphere version: 0.10.1

### Environment?

I modified original BrowseNX to adapt it for new libnx api, the modified code can be found on https://github.com/amicuchu/BrowseNX
Test was executed using emummc
Vanilla atmosphere was used loaded via hekate.
No additional sysmodules has been used

### Additional context?

If the program is executed without press any button, web applet will be invoked.
If you press the button - while the message text is shown, wifi auth applet will be invoked

fincs, 2019-12-27:
Try copying config_templates/system_settings.ini to config/system_settings.ini, then uncommenting this line (https://github.com/Atmosphere-NX/Atmosphere/blob/master/config_templates/system_settings.ini#L43) and bumping it up to a larger number.

amicuchu, 2019-12-28:
I tried with size 0x1B400000 and keeps doing the same issue. I rebooted the system to be sure that config changes are applied correctly

SciresM, 2019-12-28:
0x1B4... is wildly larger than the amount actually reservable, as a note regardless of other diagnostics

amicuchu, 2019-12-30:
The issue produced by an old hbloader. Sorry for the incovenience

closed 2019-12-30 by amicuchu