Atmosphere-NX/Atmosphere #783: Unable to start any game, getting the "Unable to start software. Return to Home Menu.."

EddyCR, 2019-12-30:
## Bug Report

### What's the issue you encountered?
After upgrading to 9.1.0 and doing a fresh install of atmosphere i 0.10 I'm not able to open any game, either already installed games or a new installations, i get the "Unable to start software. Return to Home Menu and try again?

I used to run Reinx  with 8.1.0, I removed all the reinx files and did the update to 9.1.0 before installing</br>

### How can the issue be reproduced?
-Boot to RCM
-Launch Hekate
-Boot Atmosphere through fusèe-primary
-Once in the menu try to start any game

### Crash Report
NO</br>

### System Firmware Version
9.1.0|ASM 0.10.1/S</br>

### Environment?
Bootloader: Hekate
Release: 0.10.1 (Official)
Additional Kips/Sysmodules: N/A
Homebrew Installed: N/A

### Additional context?
- Additional info about your environment: I used to run reinx with 8.1.0, the system was updated when i had the switch connected to the pc with FTP, after that by mistake I restarted the switch and then it updated to 9.10, after that I clear all the reinx files and did a fresh install fo hekate and atmosphere.
-I'm a mac OS, the sd it's on FAT32.
-I already tried both hekate's archive bit fixers.

Thanks for the help

drofdoom, 2019-12-30:
You need signature patches not included in Atmosphere.

SciresM, 2019-12-30:
Re-download your games from the eShop.

closed 2019-12-30 by SciresM

EddyCR, 2019-12-30:
I'm not gonna get ban from doing that? I also did a fresh install of some games with Goldleaf but those arent opening either.