Atmosphere-NX/Atmosphere #786: crash in Atmosphere - not stable

demetom, 2020-01-02:
Hello,

I don't know exactly what to do with theses issues. I already tried so many things in order to solve this but I really don't know what to do now. I am even thinking of selling my unpatched switch as I am so frustrated now...

I have an unpatched Switch: XAJ7001373 OFW: 9.1
I installed Atmosphere 10.1 and Hekate 5.1.1 with the patches from Joonie86.
I managed to partition my SD card with Tegra Explorer and I managed to create en EMUNAND.
I also managed to launch Hekate and install NSPs with Goldleaf.
However my system appears to be unstable. Randomly I have some frozen screens when I am playing, or some crashes with 2002-2002 error code and 2002-3502 code.
I tried the tool Archive Bit of Hekate and I always have some archive bit fixed (typically 60 in the Nintendo folder) even if I use the tool several times in a row as if it wasn't correcting the archive bits (don't know if it is normal)
I tried different SD cards (SANDISK ULTRA and SAMSUNG EVO+) that appear to be genuine, I checked them with h2testw...
I am kind of desperate. If someone could help me in order to get a stable system, I would really appreciate. (For one of my friend, it works really well and it is stable)
One of my thought is that my sysnand is dirty. I used a RCM loader and I had a very weird message (like a virus message when it loaded the payload the very first time)
Is it possible that with a dirty sysnand it could create some issues like that.
I know that I should have made a clean copy of my sysnand from the beginning but it appears that the copy I made at the very beginning is not good because when I tried to restore it, I had an error message.
I used Haku33 once in order to clean my sysnand but I am wondering if it is enough to clean it completely. I was able to connect to the eshop and to download demo games from it without being banned.
Please, if anyone has an idea on what to do in order to get a stable system working, I would really appreciate. Thanks a lot

[image]
[image]
[image]
[image]
[image]

SciresM, 2020-01-02:
> install NSPs with Goldleaf.

Piracy is not supported.

>2002-0002

"File not found".

>2002-3502

"Mmc device isn't active".

No idea why `es` would get this error.

I don't think this is an *atmosphere* issue from what I'm hearing.

Does issue occur using completely stock Atmosphere release build, booting with fusee and not hekate? Closing because it seems like not-ams-problem, talk to m4xw about nand weirdness maybe.

closed 2020-01-02 by SciresM

geraldFF, 2020-03-21:
Hi! I have the same issue, did you found out the solution OP? Thank you!

demetom, 2020-04-07:
Nope sorry, I didn’t find any solution for this but I think it is a problem of the Switch itself. 
(Maybe the card reader but I am not sure)
And it worked perfectly with an other Switch

Neno2113, 2020-07-02:
I have the same problem and I think is card reader of the Switch

SimoneCupo, 2022-03-19:
> I have the same problem and I think is card reader of the Switch

Did u fix it?