Atmosphere-NX/Atmosphere #788: Title override only overrides with hbl.nsp

melsbacksfriend, 2020-01-04:
## Bug Report

### What's the issue you encountered?
I am trying to make it so that Pokémon Sword is overridden by ulaunch, which is in .nsp format but Super Smash Bros. Ultimate still loads hbl.nsp.  When I try that, Pokémon Sword still loads hbl.nsp.  here's my .ini.  
`[hbl_config]
; Program Specific Config
; Up to 8 program-specific configurations can be set.
; These use `program_id_#` and `override_key_#`
; where # is in range [0,7].
; program_id_0=010000000000100D
; override_key_0=!R
program_id_1=0100ABF008968000
override_key_1=R
path=atmosphere/ulaunch.nsp
program_id_2=01006A800016E000
override_key_2=R
path=atmosphere/hbl.nsp

; Any Application Config
; Note that this will only apply to program IDs that
; are both applications and not specified above
; by a program specific config.
override_any_app=false

[default_config]
override_key=!Y
cheat_enable_key=!Y                                                                                        `

### How can the issue be reproduced?
by editing the file the way I did and moving ulaunch to /atmosphere and renaming it to ulaunch.nsp

### Crash Report
none

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: official
0.10.2
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: none

### Additional context?

- Additional info about your environment: none

SciresM, 2020-01-04:
Piracy is not supported. This feature is for HBL, not warez.

SciresM, 2020-01-04:
Oh, this isn't actually a warez issue, misread. Still, issue should be closed. Sec while I type that up

SciresM, 2020-01-04:
The feature at present isn't for overriding different titles with different contents. It's for hbl, which has a bunch of first class/special support in atmosphere. Lots of components behave differently when talking to a HBL process vs a non-HBL process. As a result, only one "path" entry is supported, since the user should only have one copy of hbl.

Also, ulaunch is a home menu replacement - I don't think you even can launch it over an application(?)

closed 2020-01-04 by SciresM

TwoLeaves, 2020-01-20:
A user might have an additional copy of hbl if they have modified hbl to load an executable other than hbmenu. I personally have multiple hbl copies which I've patched to load different applications and would appreciate being able to target these copies individually.

SciresM, 2020-01-20:
Multiple copies of/modified copies of hbl aren't supported by Atmosphere (certainly not in GitHub issues, anyway).

fincs, 2020-01-20:
Loading anything other than hbmenu.nro as the default (and first) application is not supported by nx-hbloader. Additionally, the NRO environment exposed by nx-hbloader depends on hbmenu being the first application to run due to several applet related reasons. Running anything that's not hbmenu results in undefined behavior.

TwoLeaves, 2020-01-20:
Ah, just dumb luck then that I've been able to run e.g. Retroarch in this manner until now. From the Github description of hbloader ("Host process for loading Switch homebrew NROs") I mistook it for a generic homebrew application loader.

Exvat, 2020-10-26:
I actually have a use case for this that's not simply a modified hbl. I replaced the album viewer with an applet that allows easy access to the browser with the ability to enter URLs. I currently use fs-mitm to simply override the exefs.nsp of the album viewer, but I'd love to be able to use an override key instead so I can still access the album viewer proper.

Skonikol, 2020-12-10:
> I actually have a use case for this that's not simply a modified hbl.
> 
> I replaced the album viewer with an applet that allows easy access to the browser with the ability to enter URLs. I (used to) use fs-mitm to simply override the exefs.nsp of the album viewer, but I'd love to be able to use an override key instead so I can still access the album viewer proper. No nx-hbloader code is involved in this.

The key override is equal to L by default. It should work... Please tell me how to create exefs.nsp for mitm target homebrew?

Exvat, 2020-12-10:
Here's a zip with the source code and the exefs.nsp: [web.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5670698/web.zip)