Atmosphere-NX/Atmosphere #789: Error preventing atmosphere to boot

CommicKick69ing, 2020-01-05:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

A fatal error occurred when ruming Atnosphere.
 Title ID: 00ff747765616bff
 Error Desc: std::abort() called (Oxffe) 
Report saved to atmosphere/fatal errors/report_00000000005Z158a,bin Press POMER to reboot

### How can the issue be reproduced?

[ * ]</br>
I have no idea. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-01-05:
Please upload the error report.

friedkeenan, 2020-01-05:
That looks like sys-tweak, which is bundled with BrowseNX and is known to cause these aborts. Deleting `atmosphere/contents/00ff747765616bff/` *should* fix this issue

SciresM, 2020-01-06:
Oh, yeah, title-id is a custom sysmodule, so not an ams issue.

closed 2020-01-06 by SciresM