Atmosphere-NX/Atmosphere #790: dmnt_cheats_enabled_by_default still not work

sakatsuku, 2020-01-06:
Atmosphere 0.10.2
I set dmnt_cheats_enabled_by_default to u8!0x0, but still not work
It happens on 0.10.1 and I submitted this issue too

herman23a, 2020-01-06:
you must move your file system_settings.ini from /atmosphere/    to    /atmosphere/config/

dont know why system_setting.ini in /atmosphere/ in the first place.

SciresM, 2020-01-06:
1) Use issue templates

2) 0.10.0 moved configuration settings per above, see release notes.

closed 2020-01-06 by SciresM

sakatsuku, 2020-01-07:
> you must move your file system_settings.ini from /atmosphere/ to /atmosphere/config/
> 
> dont know why system_setting.ini in /atmosphere/ in the first place.

thanks，I moved system_settings.ini to /atmosphere/config/，and delete "; " in front of dmnt_cheats_enabled_by_default, it works