Atmosphere-NX/Atmosphere #793: Atmosphere In Mariko Switch 2019 Edition (patched unit)

snuckdev, 2020-01-09:
## Feature Request

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ If you do not have the information needed for a section, replace its contents with "Unknown". ]</br>
[ Lines between [ ] (square brackets) are to be removed before posting. ]

[ Please search for existing [feature requests](https://github.com/Atmosphere-NX/Atmosphere/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22features%2Ffeature-request%22) before you make your own request. ]</br>
[ Duplicate requests will be marked as such and you will be referred to the original request. ]

### What feature are you suggesting?
#### Overview:
- [ Include the basic, high-level concepts for this feature here. ]</br>
Will Atmosphere developers team find a way to Jailbreak the Mariko Switch?

#### Smaller Details: 
- [ These may include specific methods of implementation etc. ]</br>
N/!

#### Nature of Request:  
[ Remove all that do not apply to your request. ]
- Addition

### What component do you feel this would best fit within?
- [Fusée](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's custom bootloader.</br>
- [Exosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Fully-featured custom secure monitor.</br>
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>
- [**Thermosphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Atmosphère's emuNAND implementation.</br>
- [**Troposphère**](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
  - Application-level patches to the Horizon OS.</br>
All of them

[ Note: **Bolded components are not implemented** or are still at the prototyping phase. ]

### Why would this feature be useful?
[ If this is a feature for an end-user, how does it benefit the end-user? ]</br>
[ If this feature is for developers, what does it add to Atmosphère that did not already exist? ]</br>

The people that bought the new Switch will have a way to jailbreak it.

deadair3210, 2020-01-09:
You realise there has to be an exploit first, right?

SciresM, 2020-01-09:
Atmosphere is a post-exploitation payload, not an exploit.

closed 2020-01-09 by SciresM