Atmosphere-NX/Atmosphere #794: cannot run game card even with nogc = 0 in BCT.ini

huynht12, 2020-01-09:
not sure why after updating to the latest atmosphere version, i keep getting "game card could not be read". Ive encountered this before and have added "nogc = 0" to the BCT.ini file, yet still getting the error.

alexgesti, 2020-01-09:
I do'it that too and dosen't work. For worst things, now i CAN'T log into any "downloaded" game or program from hb (sometimes i can log but dosen't work, just crashes)

huynht12, 2020-01-11:
Ok I've figured it out, since im using the packaged version from Kosmos, they included hackate in there too. This included a file "Hekate_ipl.ini" there's a line the say "autonogc" that was set to 1, this needed to set to 0.

closed 2020-01-11 by huynht12

rushairer, 2020-05-24:
10.0.2 with 0.12.0 atmosphere

Case 1:

Hekate_ipl.ini  ->    autonogc = 0
BCT.ini  ->  nogc = 0
Can run game card, but CAN'T run "download" game

Case 2:

Hekate_ipl.ini  ->    autonogc = 1
BCT.ini  ->  nogc = 0
CAN'T run game card, but can run "download" game