Atmosphere-NX/Atmosphere #795: Error 2168-0002 ox4a8

Maykoncorreia, 2020-01-10:
### What's the issue you encountered?

[ After updating atmosphere to 9.1 by choidujoirNX, I get this error seconds after boot. ]</br>
[ N/A ]</br>
[ N/A ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ After injecting the payload the system gets main screen error after 5s]</br>

### Crash Report

https://gist.github.com/Maykoncorreia/18088b5cf01add6d2910d68409ad00b8

### System Firmware Version

9.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe
- Official release or unofficial build: official
  - [ Official release version 0.10.2 (or) unofficial build ]
 
- Homebrew software installed: [latest ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

CompSciOrBust, 2020-01-11:
Did you have a theme installed?

SciresM, 2020-01-12:
Remove ldn.mitm, which is crashing and not an atmosphere component.

closed 2020-01-12 by SciresM