Atmosphere-NX/Atmosphere #796: Error on boot

Boucingblade, 2020-01-10:
Bug Report
What's the issue you encountered?
The following error apeers when trying to boot from Hekate, atmosphere (sysnand)

A fatal error occurred when running Atmosphére.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Did you make any changes related to Atmosphère itself?
No

How can the issue be reproduced?
Boot to RCM and launch Hekate and then Boot Atmosphere primary or emunand

Crash Report
Not a crash report but a Fatal Error maybe the same error: https://github.com/Atmosphere-NX/Atmosphere/issues/752

System Firmware Version
8.1.0

Environment?
Bootloader: Hekate
Release: idk
Additional Kips/Sysmodules: N/A
Homebrew Installed: N/A

Additional context?
This is happening after trying to upgrading from hetake 5.0.2 to the 5.1.1.
Already re-uploaded archives, Windows user, sd card on fat32.

SciresM, 2020-01-10:
Please upload the fatal error report. Help can't be given without it.

EddyCR, 2020-01-10:
Have you tried the same solutions given on #752, running the both hekate's archive bit fixer?

Boucingblade, 2020-01-10:
> Have you tried the same solutions given on #752, running both hekate's archive bit fixer?

already tried.

Boucingblade, 2020-01-10:
> Please upload the fatal error report. Help can't be given without it.

How? what kind of upload?

Boucingblade, 2020-01-10:
> Please upload the fatal error report. Help can't be given without it.

[image]

HookedBehemoth, 2020-01-10:
> Report saved to /atmosphere/fatal_errors/report_[...].bin
Send that file.

SciresM, 2020-01-10:
Please upload the file that is being saved to /atmosphere/fatal_errors, yes

Boucingblade, 2020-01-12:
The Error file

Boucingblade, 2020-01-12:
> Please upload the file that is being saved to /atmosphere/fatal_errors, yes

The error file, isnt supported:
https://drive.google.com/file/d/10PWQVWmMcvByRhD54fwqzZo5EHp9BK-0/view?usp=sharing

SciresM, 2020-01-12:
Your PRODINFO partition is corrupted. Please seek assistance in the reswitched discord if needed, but this isn't actually an atmosphere bug.

closed 2020-01-12 by SciresM