Atmosphere-NX/Atmosphere #797: Improve docs

macabeus, 2020-01-11:
## Feature Request

I'm studying this amazing project, and I think that we could improve the documentation.

### What feature are you suggesting?

- [`fuse.md`](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/components/fusee/fusee.md)
  - missing explaining what is `fusee-mtc`
- [`exosphere.md`](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/components/exosphere.md)
  - I think that the `upcoming software exploits` already is obsoleted. Was it really had a big role with Jamais Vu and Déja Vu?
  - `Exosphère currently only contains one extension` is obsoleted. On [0.8.3 released](https://www.reddit.com/r/SwitchHaxing/comments/ajf0u0/atmosphere_083_released/) a new extension was added. Are there others extensions?
  - what is the purpose of the folders `lp0fw`, `rebootstub` and `sc7fw`?
- [`stratosphere.md`](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/components/stratosphere.md):
  - would be nice to say that `fs_mitm` is a module inside of a bigger module called `ams_mitm` that it has many others mitm modules
- [`thermosphere.md`](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/components/thermosphere.md) and [`troposphere.md`](https://github.com/Atmosphere-NX/Atmosphere/blob/master/docs/components/troposphere.md):
  - both documents say that these features isn't implemented yet, but there are some code on respectively folders... is it up to date?

I would love to help in this improvements, but I'm still very beginner in this project. If someone could help me to improve these documentations, I would be very happy =]

lordfriky, 2020-01-21:
I would like to help on this too, I'm writing a spanish blog post explaning all about Atmosphère in depth, since no other web seems to care so much about this details.

One specific question that has been in my mind for some days now is: Is *emuMMC* cosidered *Thermosphère* now? If not, will *Thermosphère* will be worked at some point? Or that idea is abandoned now?

hexkyz, 2020-01-27:
Sorry for the delayed answer. I'm assigning this issue to myself as I'll go over and update all the documentation. Please keep using this issue to suggest any further changes.

hexkyz, 2020-05-01:
Closing this issue as the documentation has been fully updated.

closed 2020-05-01 by hexkyz