Atmosphere-NX/Atmosphere #802: Unknown pkg1 version Failed to boot HOS

BladeStrde, 2020-01-19:
### What is the issue:

By starting the EmuMMC from Hekate I receive an error stating unknown pkg1 version failed to launch HOS

### How it can be reproduced
Open hekate from RCM -> Launch -> CFW (EMUMMC)

### What actions have I taken

- Redownload everything including sig patches
- Use Fusee-primary (results in a black screen) 

I really am lost at the moment and can supply extra information if needed

SciresM, 2020-01-19:
This is a hekate error, not an atmosphere error.

Update both atmosphere and hekate to the latest versions.

closed 2020-01-19 by SciresM

BladeStrde, 2020-01-19:
Just updated both, problem still persists

SciresM, 2020-01-19:
Seek help with hekate or in the ReSwitched support channels -- this isn't an atmosphere problem, hekate is not an atmosphere component.