Atmosphere-NX/Atmosphere #804: Class of Heroes: Anniversary Edition english romfs replacement crash

anotherswitchanon, 2020-01-21:
## Bug Report

### What's the issue you encountered?

Game crashes with replacement romfs files from english PSP game, it worked previously on any version from v0.94 and back, I'm assuming it stopped working on v0.10 as I'm on 0.10.2.

### How can the issue be reproduced?

*  Copy [english files](https://drive.google.com/open?id=1QmO1FiDiGHmUsSzqyGSETi9ev2oNhRwC) to sdmc
* Install game, 01003AF006A6C000 is the titleid.
* Run
* Crash

### Crash Report

https://gist.github.com/anotherswitchanon/e1e6b45de5fa3696e9d47eafcb18ae6c

### System Firmware Version

9.1.0

### Environment?

- hekate v5.1.1
- Official release v0.10.2
- sys-ftpd-light & sys-clk sysmodules installed

### Additional context?

- Additional info about your environment:
- N/A

SciresM, 2020-01-21:
:/ @ debugging 32-bit games.

Please trigger the crash again after disabling usermode exception handling in /atmosphere/config/BCT.ini (ignore the warning there about my not providing support in this case), and then upload the resulting crash file?

SciresM, 2020-01-21:
Also, please verify whether issue occurs without custom system modules (in addition to the above test).

anotherswitchanon, 2020-01-21:
Sure thing, here's the gist from it.
https://gist.github.com/anotherswitchanon/99d3d4421c8ddbcae5539f711a00621d

Tested again with all sysmodules disabled and still crashes unforunately, the gist has sysmodules disabled.

SciresM, 2020-01-21:
This crash also has user exception handlers enabled. Could you double check that you changed the setting?

You may need to boot via fusee-primary for that change to take effect -- I'm not sure if (or how) hekate exposes it.

anotherswitchanon, 2020-01-21:
Hopfully it's disabled this time, sorry about that.
https://gist.github.com/anotherswitchanon/4743441986dfde33031fc97fa49ad460

SciresM, 2020-01-21:
Same thing. Are you booting with hekate?

anotherswitchanon, 2020-01-21:
Ah, that's why. Give me a sec.

anotherswitchanon, 2020-01-21:
https://gist.github.com/anotherswitchanon/aa8bd9b2c241e1dea150af1df5e8c59c
Did it work?

SciresM, 2020-01-21:
Heh, still nope. Could you paste the contents of your BCT.ini so I can validate the setting?

btw, make sure you replaced sept/payload.bin with fusee-secondary.bin temporarily (as it is freshly extracted from the atmosphere zip). If you don't, you'll be booting with hekate despite launching fusee-primary lol.

anotherswitchanon, 2020-01-21:
Oh, I didn't know that, give me a sec.

Here's my BCT.ini.
https://gist.github.com/anotherswitchanon/116f23dde063db370a028b73cc4a7a32

SciresM, 2020-01-21:
Yeah, that looks good.

On a successful boot via fusee you'll see the blue atmosphere splash screen for a few seconds. (sept screen -> blue atmosphere screen -> black and white atmosphere screen -> nintendo switch screen)

SciresM, 2020-01-21:
Yeah, that's a usable log, thank you -- will investigate tonight.

anotherswitchanon, 2020-01-21:
No problem, thanks so much.

CTCaer, 2020-01-21:
Btw, hekate needs a key in the selected boot section for that.

[image]

Example:
```ini
[Atmosphere Vanilla]
fss0=atmosphere/fusee-secondary.bin
nouserexceptions=1
```

SciresM, 2020-01-21:
Gotcha, good to know for future debugging.

SciresM, 2020-01-22:
@anotherswitchanon I'm not sure I'm convinced this is atmosphere and not the game/your mod.

Testing reveals that the game crashes if either of the files "data/texture/title.enc" or "data/texture/texture.enc" are present on SD.

The RomFS metadata built for the game is valid when the game crashes, and is built successfully.

Further, the game works redirecting unmodified copies of those files to the SD.

I observe that while the other files included in your mod are roughly the same size as the files present in the actual RomFS, there is a *huge* disparity in file size between your mod's title.enc and texture.enc.

In particular, your texture.enc has size 488 KB and your title.enc has size 122 KB.

The versions of these files in the romfs have size 26.3 MB and 16.8 MB respectively.

If you really believe that this mod worked on atmosphere 0.9.4, could you please run the game under atmosphere 0.9.4 and send me the "romfs_metadata.bin" generated in the relevant program directory afterwards?

anotherswitchanon, 2020-01-22:
Is there a quick way to get atmosphere 0.94 to work because it's not working, I keep getting pkg2 decryption failed when it was fine on 0.10.2?

SciresM, 2020-01-22:
You'll need to boot into a firmware version that 0.9.4 supports, I suppose. Make an emummc and downgrade it?

anotherswitchanon, 2020-01-23:
Ah, you're right, I made a mistake with the patch, sorry for the trouble.

closed 2020-01-23 by anotherswitchanon