Atmosphere-NX/Atmosphere #805: Panic occured

marvimarv, 2020-01-23:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

When I try to launch Atmosphere or Kosmos I always get this error message:

> Pannic occured while running Atmosphere
> 
> Title ID: 010041544D530000
> Error Desc: std::abort() (0xFFE)

I upload the bin file
[report_05601735.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4101917/report_05601735.bin.zip)

### How can the issue be reproduced?

Even on a clean SD card when I launch into the CFW i get this error.

### Crash Report

uploaded as attachment

### System Firmware Version

9.1.0

### Environment?
hekate 5.1.1

clean atmosphere version with no other software

### Additional context?

N/A

SciresM, 2020-01-23:
Fix up your archive bits. Are you using macOS?

This error report corresponds to a failure to create/open the automatic backup files on boot, which happens when archive bits prevent access to the relevant directory (as HOS assumes it's a file).

closed 2020-01-23 by SciresM

marvimarv, 2020-01-23:
Heck yeah that did it! I used both of the archive fixing things on Hekate. Thanks!

abiem, 2020-01-30:
this is amazinggggg THANK YOU for this tip!!!!!!! took me hours to figure it out and this fixed it!