Atmosphere-NX/Atmosphere #806: Can't boot into atmosphere from a cold boot

BruceHan1998, 2020-01-28:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
</br>

</br>
</br>
N/A</br>
N/A</br>
</br>

### What's the issue you encountered?

I installed kosmos v15.1 on my switch, then I pulled out the mircosd card to insall a game, which caused the console shut down. Then I was noat abl to boot into atomosphere anymore even after injecting the payload agin, it wiould just boot into the official system</br>
N/A</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
Just like I said eariler, I pulled out the sd card, and put it back in, then I tried to boot the switch, but it would only boot into the offical system.</br>

### Crash Report

N/A</br>
N/A</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

9.1.0</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate_ctcaer_15.1.1.bin
- Official release or unofficial build: 
  - Kosmos v15.1
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: checkpoint, goldleaf

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-01-28:
1) Atmosphere does not support booting without an SD card
2) Installing games is not supported.

closed 2020-01-28 by SciresM

BruceHan1998, 2020-01-28:
> 1. Atmosphere does not support booting without an SD card
> 2. Installing games is not supported.

As I stated in the original post, I put the sd card back in, and then tried to boot, but it didn't work

drofdoom, 2020-01-28:
Its not ofw check your system after booting you may need just to install
sig patches not included in Atmosphere