Atmosphere-NX/Atmosphere #808: Launching eshop through games freezes console.

smrtg, 2020-01-30:
## Bug Report

### What's the issue you encountered?

Tried to launch the eshop through either Super Smash Bros Ultimate's shortcuts (to buy dlc) and Pokemon Shield's shortcuts (for the same reason) 

No changes were made by me to Atmosphere 

### How can the issue be reproduced?

Launch any game that can take you to the eshop for (mostly) dlc
Press the eshop shortcut 
Done, the eshop loading screen (the orange bars) should be frozen and the console also completely frozen too. 

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
hekate
- Official release: 
  - 0.10.2|S
- Do you have additional kips or sysmodules you're loading:
sys-con
sys-ftpd-light
sys-clk
- Homebrew software installed: 

RetroArch
Atmosphere Updater
Browser (Internet browser)
Checkpoint 
EdiZon
Ftpd
Gamcard Installer
Goldleaf
HB App Store
Kosmos Toolbox
Kosmos Updater 
NXThemes Installer 
Reboot To Payload

ELY3M, 2020-01-30:
Can you buy thru the normal eshop?   I have no issues with using normal eshop.   I do have both of the games on game carts.  I never tried to access eshop inside the games.

ELY3M, 2020-01-30:
I just tested in pokemon sword and I can't reproduce the issues. it just forward to the normal eshop.

ELY3M, 2020-01-30:
are you running 90dns?   that will stop eshop from working.

smrtg, 2020-01-30:
Hey, tha ms for the reply. I can launch and buy from the home menu. I have both games digitally, don't know if that matters. And I'm not using 90dns. I can send a vid of it happening if you want. I've tried to launch eshop a total of ě times through an app, and all three is froze

SciresM, 2020-01-30:
Does error occur using stock atmosphere without custom sysmodules?

smrtg, 2020-01-31:
> Does error occur using stock atmosphere without custom sysmodules?

Haven't tried it yet. I'll try and post the result

fennectech, 2020-02-17:
Ive got the same issue here   ill grab an sd card and do it on plane jane atmosphere.

SciresM, 2020-02-17:
@fennectech "But yes this also happens on raw atmo" disagree.

That crash is from a service client being closed (terminated) in the middle of mitm session ACK. This can only happen if the process itself is terminated externally, which can never happen using stock atmosphere.

Also, the report you linked is unrelated to the freeze mentioned in the issue.

fennectech, 2020-02-18:
Ive got an sdcard with plain atmo (fusee-primary chainloaded with hekate) that exibits the hang. (not the crash)    i wasn not refering to that crash for the happening on atmo.

SciresM, 2020-02-18:
The hang as diagnosed so far only occurs with additional sysmodules present -- it's memory pool stuff.

It doesn't happen with stock atmosphere.

ELY3M, 2020-02-21:
note:  I am on stock atmosphere with sys-ftpd-light.

hmax42, 2020-04-14:
remove the sys-ftpd-light, that's a sysmodule

SciresM, 2020-04-20:
Still can't reproduce this using stock atmosphere. Closing.

closed 2020-04-20 by SciresM