Atmosphere-NX/Atmosphere #809: Keep showing Error Code: 2168-0002 (0x4a8) in my Nintendo Switch

kobey88, 2020-01-30:
## Bug Report

### What's the issue you encountered?

Sorry I'm a noob for programming or gaming. I have bought a second-hand switch which was installed Atmosphere already. I have just press update when I open a game and connected in my home WiFi, and the NS shows error. 

I have updated to 0.10.2 master from 0.9.x

### How can the issue be reproduced?

When the NS showed me the error first time, I turned off the NS and I can't turn it on. It was showed fatal error nx boot unable to identify package on that time. After that, I tried to update Atmosphere to 0.10.2 master. The NS have been turned on but it shows me another error code which is 2168-0002 (0x4a8) so I dun know how should I do next step. 

### Crash Report

[01580398994_4200000000000010.log](https://github.com/Atmosphere-NX/Atmosphere/files/4134897/01580398994_4200000000000010.log)

### System Firmware Version

9.1.0

### Environment?

hekate
Sorry I have no idea that is it Official release or unofficial build

kobey88, 2020-01-30:
Opps sorry...for the part of environment, should be 0.10.2 from official release

shchmue, 2020-01-30:
Update or disable `ldn_mitm`. It's in `atmosphere/contents/4200000000000010` on SD.

kobey88, 2020-01-30:
> Update or disable `ldn_mitm`. It's in `atmosphere/contents/4200000000000010` on SD.

I don't have this file :(

shchmue, 2020-01-30:
that means when you updated atmosphere you possibly didn't inject a new payload, so then it's probably in `atmosphere/titles/4200000000000010`

kobey88, 2020-01-30:
> that means when you updated atmosphere you possibly didn't inject a new payload, so then it's probably in `atmosphere/titles/4200000000000010`

But however I don't have `4200000000000010` in `atmosphere/titles` as well. How can I get the `ldn_mitm`? :')

SciresM, 2020-01-30:
Delete custom sysmodule, if you really can't locate it then just delete atmosphere and extract clean latest release

closed 2020-01-30 by SciresM

kobey88, 2020-01-30:
> Delete custom sysmodule, if you really can't locate it then just delete atmosphere and extract clean latest release

Thanks for your help and the problem solved.