Atmosphere-NX/Atmosphere #812: Black Screen after updating to last firmware and last atmosphere/Hekate

combluotion, 2020-02-03:
## Bug Report
### What's the issue you encountered?

After updating to last switch firmware, updated atmosphere and Hekate, I find a black screen when entering the payload.
### How can the issue be reproduced?
Followed https://nh-server.github.io/switch-guide/extras/updating/ guide. Specifically sections:
"Updating Atmosphere", "Updating Hekate", "Updating Firmware"

I Updated to last switch firmware and proceeded to update atmosphere and hekate following above guide, sadly now when I try to launch the payload it just goes black screen after showing some atmosphere's logos

### Crash Report

N/A

### System Firmware Version

9.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
Atmosphere last release to date and hekate.
- Official release or unofficial build: 
 Release build.
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: Goldleaf, tinfoil, njourdswitch

### Additional context?

- Additional info about your environment:

I haven't got many, maybe someone that had a similar issue could overcome it?

SciresM, 2020-02-03:
This issue only occurs when you have non-updated ams/hekate (and in particular when you don't have correct sept). Re-extract latest release builds.

closed 2020-02-03 by SciresM

combluotion, 2020-02-03:
I'm using both Hekate and Atmosphere's last version. (With the latest sept folder) maybe there is some corruption on my switch firmware?

SciresM, 2020-02-03:
No, you're just not actually using updated ams/hekate even though you think you are.

SciresM, 2020-02-03:
Make sure you're not using auto updaters -- they're not reliable

combluotion, 2020-02-03:
I downloaded both releases from here:

https://github.com/Atmosphere-NX/Atmosphere/releases/tag/0.10.2
https://github.com/CTCaer/hekate/releases/tag/v5.1.1

I replaced the files on the SD card and put the fusee-primary.bin on the bootloader/payload folder

I Injected the Hekate_ctcaer_5.1.1.bin with the ns-atmosphere updater.

Do you think I am missing something else? 

Thank you for your work. I really appreciate it.

SciresM, 2020-02-03:
"ns-atmosphere" isn't associated with us in any way.

I don't really know what you're talking about wrt bootloader/payload folder. In any event, this doesn't seem like a bug -- seek support in a support channel.