Atmosphere-NX/Atmosphere #813: Fatal error ocurred when running Atmosphére #010041544d530000

NicolasCLMendes, 2020-02-05:
## Bug Report [Fatal Error Ocurred when running Atmosphére.]

Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)

Press POWER to reboot.

### What's the issue you encountered?

I was in version 9.0.1, but when I had updated to the 9.1.0 using ChoiDujourNX, the Atmosphere stop working, I formated my SD (from ExFat) to FAT32 but that not worked,I've tried to use chmod and chflags, but nothing works. But the Update was sucessfully, because I tested using sxos, but I prefer Atmosphere...</br>

I not changed anything in the atmosphere folder and a have tried a clean installation with only Atmosphere.</br>

### How can the issue be reproduced?

Just injecting the Atmosphere payload 10, 10.1 or 10.2</br>
the issue happens after the White Icon splash screen</br>

### Crash Report

Report from fatal_errors folder:
[https://1drv.ms/u/s!AkqmmjrMDNW3kjBzgA-0J1QUB90x?e=pgxhS3](url)

### System Firmware Version

9.1.0 and Atmosphére 10, 10.1 & 10.2</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: TegraSmashGUI.exe
- Official release or unofficial build: Official release 10, 10.1 and 10.2
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: [ Appstore, Checkpoint, ChoiDujourNX, LinkUser, mmlinx, nxmtp,NX-Shell, Tinfoil, Sys-clk-Editor, Edizon]

### Additional context?

When I have tried with clean Atmosphere installation the report was the same, but the some description was in Bold and Yellow.

Ness-666, 2020-02-15:
I am having the exact same issue, the only thing I thing that I can think of is that i had removed my prod.info and since then lost my back up. Not sure if that would effect it though.

SciresM, 2020-02-15:
This is a corrupt prodinfo problem -- your PRODINFO somehow has non-ascii characters in the serial number. 

This currently isn't supported by Atmosphere.

closed 2020-02-15 by SciresM

Manuel-Flores-F, 2020-05-14:
This is usually due to an archive metadata bit used by some OS’s that Horizon simply can’t deal with. Here’s how to fix it:

Get your switch in RCM mode.
Inject Hekate payload
Tap on "Tools" on top of the hekate screen.
Tap on "Archive Bit - AutoRCM" on the right bottom.
Tap on "Unset archive bit" button. This might take a while, be patient.

Barkermush, 2020-05-17:
I'm getting this exact same problem. It was fine until I updated from 10.0.1 to 10.0.2.
I updated atmosphere, I've formatted the memory card and tried a clean install but I'm still getting this error.

I think I might have to try a full nand backup too, as I don't seem to be getting anywhere.

SciresM, 2020-05-17:
"same problem" probably not. literally any error in ams.mitm generates this symptom to user. Upload your report binary.

Barkermush, 2020-05-17:
Ah I never knew that, thanks.
I will do that now.

Barkermush, 2020-05-17:
> "same problem" probably not. literally any error in ams.mitm generates this symptom to user. Upload your report binary.

Sorry to be a pain but do you have a link to something showing how to get the 'binary report' had a look on hekate and can't see anything for that.

The only stuff I can dump to my sd card is 'console info/soc info.. Ipatches & bootrom.. fuses' etc

SciresM, 2020-05-17:
When an error occurs, a report is saved in /atmosphere/fatal_errors

Barkermush, 2020-05-17:
> When an error occurs, a report is saved in /atmosphere/fatal_errors

Thanks.
Uploaded the error report here: https://www.mediafire.com/file/aoc2jpvnxcvenih/report_0000000005fb318e.bin/file

Barkermush, 2020-05-17:
[report_0000000005fb318e.zip](https://github.com/Atmosphere-NX/Atmosphere/files/4640236/report_0000000005fb318e.zip)

Barkermush, 2020-05-17:
> When an error occurs, a report is saved in /atmosphere/fatal_errors

Wow, I have managed to sort it. After looking at some other threads, I needed to format the sd card to fat32 (need to download a 3rd part app to do this, as Win10 as no fat32 option)

Never had a problem with atmosphere and exfat from 8.00 patches up until this 10.0.2 patch. This is the first time I have needed to have my sd card in fat32 format.

Ah well, just glad I didn't have to do a full nand :D

yokanes, 2020-09-22:
Hi friends! i have the same issue . When the switch forced to update my atmosphere dosent work. I have a micro sdxc 128gb and when i inject the payload of version 0.14.4 of atmosphere, appears this text : 

A fatal error ocurred when running atmosphere.
itittle ID: 010041544d530000
error desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_00000000559655.bin

Press POWER to reboot.

[report_0000000005996855.bin.zip](https://github.com/Atmosphere-NX/Atmosphere/files/5260140/report_0000000005996855.bin.zip)

I though is a problem with my micro sdxc 128 gb and decided bought a new micro sd xc of 256 gb and make the same steps for the new micro sdxc of 256 gb and the same issue appears again. then i test the same steps with a micro sdhc of 2gb and the issue dont appear.

Can u help me?

SciresM, 2020-09-22:
@yokanes run the archive bit fixer in hekate.

yokanes, 2020-09-22:
How can i run it the archive? i dont know what is hetake

yokanes, 2020-09-22:
100% completed the fixed. Problem solved !! 

U are the f***ing master !!

lordelan, 2021-05-02:
The archive bit fixer in hekate doesn't do the job for me unfortunately. (finished with total of 0 set/unset)
I'm in an emuMMC on 11.0.0 and am getting the same error message as OP.
Please let me know what's the problem:
[reports.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6412402/reports.zip)

lordelan, 2021-05-03:
Further info:
All I did is install Fizeau (color adjustment homebrew) from the appstore nx.
It told me that it needs newest Atmosphère, so I updated Atmosphère.
That brought me to the error screen.
I deleted Fizeau and reverted back to the old Atmosphère folder (since I set up a completely new SD card I luckily had a backup of the exact content from 1 day before on my PC).
Oddly (and that's **really** odd) it still gives me that error.

SciresM, 2021-05-03:
@lordelan you have an incomplete atmosphere directory. One of fusee-secondary.bin, sept/payload.bin, or stratosphere.romfs is missing.

lordelan, 2021-05-03:
@SciresM You're right. By restoring my old Atmosphère backup I forgot to also take the "old" corresponding sept folder.

That still confuses me why the **new** Atmosphère version wasn't booting.
Are your findings that you just wrote the same for all three included log files?

SciresM, 2021-05-03:
All three reports describe identical "file not found" crashes.

lordelan, 2021-05-03:
Okay, gotta do another update later on and take care everything works out. Thanks. :)

ImanCol, 2021-07-04:
Hekate fixed this problem with their formatting systems by resizing the MicroSD.  It is recommended to have less than 900MB of files for you to restore while formatting.

nota-debug, 2021-07-10:
me too as well it happened when my little brother updated my switch and it been happening before 19.4.0 prelease of atmosphere. i tries the bitarchive method formatting it and downloading 19.5.0. I will post the report below. please help me.

nota-debug, 2021-07-11:
[report.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6796134/report.zip)

Unlashjen24, 2021-07-17:
Hi all, ive just started getting a similar error too, everything was working just fine after I updated my atmosphere and hekate to the latest versions and I was testing some tesla cheats and I was trying to make it so the cheats didn't auto activate on startup of the application and now im getting this error, I'd really appreciate the help 
[report_0000000007ad8d64.zip](https://github.com/Atmosphere-NX/Atmosphere/files/6834676/report_0000000007ad8d64.zip)

Zylder, 2021-09-11:
I have the same problem.

Hekate was successfully installed, however when launching Atmosphere it crashes on this error.

Atmosphere is at version 12.1.0

I've tried several times, but so far nothing.

masagrator, 2021-09-11:
> Atmosphere is at version 12.1.0

This is not Atmosphere version, this is HOS version

KTVX94, 2021-09-11:
[report_0000000006a8f6af.zip](https://github.com/Atmosphere-NX/Atmosphere/files/7148327/report_0000000006a8f6af.zip)

I tried updating Atmosphere, Hekate and the Switch's OS in an attempt to dump it into the emu eMMC. First time doing it so I might have made some silly mistake. In theory they should all be the latest versions (Hekate 5.6.0, Atmosphere 1.0.0 and Switch 12.1.0).

GormAuslander, 2021-11-02:
im getting this error when I launch smash bros, after updating to atmo 1.2.2 and installing the latest smash update

takoyaki99, 2021-11-02:
Please update your ARCropolis, Training Modpack, and Skyline. Those cause the issue.
Also, fill out the issue template next time please (this helps the devs help you!)

SciresM, 2021-11-02:
(also, don't comment on resolved issues from 2020).

takoyaki99, 2021-11-02:
> (also, don't comment on resolved issues from 2020).

my bad! I got an email notification for this and assumed that it was a new issue--I should've looked at the issue number.

GormAuslander, 2021-11-02:
> > (also, don't comment on resolved issues from 2020).
> 
> my bad! I got an email notification for this and assumed that it was a new issue--I should've looked at the issue number.

You are a saint, and I apologize for the incorrect error report

PixLAK, 2022-04-24:
Hello, I'm also getting this exact error. running AMS 1.3.1 Hekate 5.7.2 and firmware 14.1.1. The error only shows up when I launch Diable downloaded from tinfoil. When I enter the game it says the save is from a new version of the game and then when i switch users it crashes and brings this up.
[report_000000164cbac037.zip](https://github.com/Atmosphere-NX/Atmosphere/files/8550572/report_000000164cbac037.zip)

BeckysFootSlave, 2022-04-25:
@PixLAK piracy is NOT supported!!!

SciresM, 2022-04-25:
@BeckysFootSlave commenting as a non-maintainer on this stuff on closed issues from two years ago is really not necessary. I can handle it.

BeckysFootSlave, 2022-04-25:
Sorry!
Didn't looked at timestamp!
Deleted

AndyII-II, 2024-07-16:
I am having the same error. Archive fix doesn't help.

Panic occured while running atmosphere.
Title id: 010041544D530000
Error: std: : abort <0xFFE>
Report saved to atmosphere/fatal_errors/report_03d34dfe9.bin
Press POWER to continue.

I don't see 03d34dfe9 file in fatal errors folder, so there is all i have
[ZIP_-_WinRAR.zip](https://github.com/user-attachments/files/16247388/ZIP_-_WinRAR.zip)