Atmosphere-NX/Atmosphere #814: Allow replacing of HTML document for specific titles

Ereza, 2020-02-05:
## Feature Request

### What feature are you suggesting?
#### Overview:
Add the ability to replace the HTML document files for a title.

#### Smaller Details: 
Currently the loader only allows the user to replace the HTML document files if launching in homebrew mode. Adding a MITM implementation (like the one for romfs) for the HTML documents would allow for modification of game help or manuals (such as in Mario Kart 8 Deluxe).

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
This would allow game mod developers (or fan translators like me) to modify the manual for games that use web applets to display it. Currently I am translating Mario Kart 8 Deluxe and I can translate all the game text thanks to fs.mitm, but I can't translate the manual because it's not possible to replace it.

SciresM, 2020-02-25:
This is half-implemented, now. The filesystem is *completely* redirected when `manual_html` directory is present under program specific dir.

Ideally we implement a kind of layered filesystem, which falls back to the real one. Will think on how best to do this, but this should make it into the next atmosphere release.

Ereza, 2020-02-25:
This is awesome news, thanks for all your work!

SciresM, 2020-02-26:
Implemented :)

commit dc140406 (2020-02-26): fs.mitm: implement layered html redirection (closes #814)
closed 2020-02-26 by SciresM