Atmosphere-NX/Atmosphere #815: SFDNSRES sysmodule with hosts file

puddydoo, 2020-02-07:
## Feature Request
### What feature are you suggesting?
#### Overview:
- Replace the [sfdnsres](https://switchbrew.org/wiki/Sockets_services#sfdnsres) sysmodule so that it can use a hosts file to resolve domain names, in addition to the DNS server.</br>

#### Smaller Details: 
- The hosts file could be used to block Nintendo's servers much like 90DNS, but it would probably be blank by default.</br>
- Hosts file is prioritized over the DNS server.</br>
- Don't connect to the DNS server if the hosts file is missing or corrupted.

#### Nature of Request:  
- Addition

### What component do you feel this would best fit within?
- [Stratosphère](https://github.com/Atmosphere-NX/Atmosphere#components)</br>
    - Custom system modules.</br>

### Why would this feature be useful?
This would work better than 90DNS when used with emuNAND because it would work with every connection and not compromise speed, or work in conjunction with another DNS server. Some users also might have other uses for a hosts file besides blocking Nintendo's servers.

fincs, 2020-02-07:
Instead of replacing it, I think we could mitm it.

SciresM, 2020-02-07:
sfdnsres is hosted by the bsdsockets sysmodule, which is big and complicated and would require reimplementing all of the sockets APIs to do what you suggest.

It might be a good target for a mitm, but there's no need for atmosphere to provide that -- if someone is interested, they can write a proof of concept sysmodule that mitm's sfdnsres as described. If it works well and is written in stratosphere's code style, then the author of such a sysmodule is welcome to talk to me about merging it into mainline atmosphere.

closed 2020-02-07 by SciresM