Atmosphere-NX/Atmosphere #816: Error 0x101

MudkipBanana, 2020-02-08:
## Bug Report

### What's the issue you encountered?

I accidentally updated my system to 9.1.0, and now on boot my switch displays the error: Panic occured while running Atmosphere.

Title ID: 0100000000000034
Error Desc: Data Abort (0x101)

### How can the issue be reproduced?

The issue cannot be reproduced.

### Crash Report

https://drive.google.com/open?id=1VaIX5xPjvBJO3rpFuPBTvZtwYZ6ITFWC

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: Hekate
- Official release or unofficial build: Kosmos
- Do you have additional kips or sysmodules you're loading: Sys-ftpd-light, emuiibo, ldn-mitm
- Homebrew software installed: Kosmos Toolbox, Kosmos Updater

### Additional context?

- Additional info about your environment:
N/A

fincs, 2020-02-08:
Use [standard unmodified Atmosphère](https://github.com/Atmosphere-NX/Atmosphere/releases/latest) without custom system modules and try again.

SciresM, 2020-02-19:
@MudkipBanana Closing, fatal is data aborting when drawing a fatal error screen (due to out of bounds framebuffer access. Your custom system font is probably not compatible with fatal.

Will consider making fatal always use actual-system-font going forwards.

SciresM, 2020-02-19:
(also, if you still get issues without e.g. custom system font etc, open a new issue).

closed 2020-02-19 by SciresM