Atmosphere-NX/Atmosphere #818: Fatal error: Failed to initialize the SD card!

boardinghousereach, 2020-02-19:
Yes, I have a hackable Switch and have hacked it many times before. However, as of right now, I am unable to load into CFW. I have not changed anything on my Switch nor my SD card. My Switch version is 9.1.0. The same version I've always been using. As of current, when I try to enter CFW (and after injecting payload), I get a message something along the line of fatal error about my SD card. I don't think there is anything wrong with my SD card, as I am still able to play my games, just not enter into CFW. Some things I've tried so far, I have dragged and dropped my SD card files onto desktop, formatted my SD card then redragged, still I get the error. I have also tried this on a different SD card, I still get the same error. Can someone please help me out?

SciresM, 2020-02-19:
Re-open using the issue templates.

closed 2020-02-19 by SciresM

ELY3M, 2020-02-19:
it look like your sd card reader may have came unseated or loose.  you have to open the switch back shell and re-tighten the sd card connector.  I have to do that few times on my old switch.    it is good idea to get ftpd and use ftp method for moving files to your sd card.

SciresM, 2020-02-19:
Please do not provide assistance on issues closed due to lack of issue templates, it discourages the correct behavior.