Atmosphere-NX/Atmosphere #819: Fatal error: Failed to initialize the SD card

boardinghousereach, 2020-02-19:
## Bug Report

### What's the issue you encountered?

I have a hackable Switch and have hacked it many times before. However, as of right now, I am unable to load into CFW. I have not changed anything on my Switch nor my SD card. My Switch version is 9.1.0. The same version I've always been using. As of current, when I try to enter CFW (and after injecting payload), I get a message something along the line of fatal error about my SD card. I don't think there is anything wrong with my SD card, as I am still able to play my games, just not enter into CFW. Some things I've tried so far, I have dragged and dropped my SD card files onto desktop, formatted my SD card then redragged, still I get the error. I have also tried this on a different SD card, I still get the same error. Can someone please help me out?

### How can the issue be reproduced?

I just did what I would normally do and inject payload with Tegra.

### Crash Report

I've tried two methods. First, I followed this video
https://www.youtube.com/watch?v=hkAixrEYJqo

Second, I followed this (after realizing I couldn't access CFW recently)
https://www.youtube.com/watch?v=EMcmSVKa1wc

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: Kosmos (I think; all files used in crash report)
- Do you have additional kips or sysmodules you're loading: Don'; think so, again refer to crash report
- Homebrew software installed: [ * ] JKSV

### Additional context?

- Additional info about your environment:
N/A

SciresM, 2020-02-19:
"Kosmos" <-- so you're using hekate? If so, this is a hekate issue and not an atmosphere problem.

(Though given our sdmmc driver is generally the same as hekate's, we probably have the same issue anyway).

In reality it sounds (as @ELY3M mentioned) like an sd card reader hardware problem.

boardinghousereach, 2020-02-19:
> "Kosmos" <-- so you're using hekate? If so, this is a hekate issue and not an atmosphere problem.
> 
> (Though given our sdmmc driver is generally the same as hekate's, we probably have the same issue anyway).
> 
> In reality it sounds (as @ELY3M mentioned) like an sd card reader hardware problem.

If that's the case, how would I go in fixing the issue?

SciresM, 2020-02-19:
@boardinghousereach `I don't think there is anything wrong with my SD card, as I am still able to play my games, just not enter into CFW.` ah never mind this'll be an SD card driver problem.

Talk to @CTCaer ^^

boardinghousereach, 2020-02-19:
> @boardinghousereach `I don't think there is anything wrong with my SD card, as I am still able to play my games, just not enter into CFW.` ah never mind this'll be an SD card driver problem.
> 
> Talk to @CTCaer ^^

Sorry, I am still new to this place. Is there a way to DM people? Or do I just shoot up an email provided on their profile?

SciresM, 2020-02-19:
Talk to him on discord, or wait for him to comment on this issue. I pinged him, odds are he'll post when he's awake.

(Of course, odds are pretty high he will just tell you to debug with him on discord or something, that's what I do for spooky problems).

boardinghousereach, 2020-02-19:
> Talk to him on discord, or wait for him to comment on this issue. I pinged him, odds are he'll post when he's awake.
> 
> (Of course, odds are pretty high he will just tell you to debug with him on discord or something, that's what I do for spooky problems).

I'm assuming there is a Discord server in which I can find him? Again, sorry I am new. How do I access the sever?

CTCaer, 2020-02-19:
That's the common DAT1/2/3 pin issue on a broken or disconnected sd reader connector.
Aka, forced 1bit mode. So nothing to debug other than check the connector status.

boardinghousereach, 2020-02-19:
> That's the common DAT1/2/3 pin issue on a broken or disconnected sd reader connector.
> Aka, forced 1bit mode. So nothing to debug other than check the connector status.

Sorry, I'm not sure I get what you're saying (I am not too familiar with computer/tecnology terms). At this point what will I need to do? Should I get another SD card reader, is there something wrong with my Switch or?

boardinghousereach, 2020-02-19:
Update, now I get a new message
Missing LP0 (sleep mode) library
Update your bootloader folder
Press any key

After that I pressed the power button once fast messages in red showed up something about failed SD

Then it loads to [hekate - CTCaer mod v5.1.1] menu with a warning message on the bottom reading, Nyx is missing

ELY3M, 2020-02-19:
@CTCaer is right.  I said same thing. 
Have to open your switch and tight up the sd reader connector.  https://www.ifixit.com/Guide/Nintendo+Switch+Micro+SD+Card+Reader+Replacement/113031

closed 2020-02-19 by SciresM