Atmosphere-NX/Atmosphere #820: Error 2168-0002 after ChoiDujourNX Update

kreiss97, 2020-02-20:
So I have an issue where I wanted to update to 9.1.0 from 9.0.0. I updated Atmosphere 0.10.2, hekate 5.1.1 and the sigpatch updater v0.1.2 and went to it. I have an exFAT SD card and selected the exFAT option. Everything seemed to go just fine. After it installed I selected "shutdown now!" and, after using TegraRCM, I cannot turn on my switch. It seems to load normally but after the Nintendo Switch logo it just prompts Error Code: 2168-0002 (0x4a8).
I have tried to delete the "titles" folder inside atmosphere, as I read that it could cause trouble, but it hasn't worked either.

[01582199785_0100000000000352.log](https://github.com/Atmosphere-NX/Atmosphere/files/4230709/01582199785_0100000000000352.log)

dennthecafebabe, 2020-02-20:
Not really a atmosphere problem afaict. The log you sent tells you that the failing process is "emuiibo" which is a custom sysmodule and not a part of atmosphere. Remove your atmosphere/contents/0100000000000352 folder and see if the problem persists

SciresM, 2020-02-20:
1) please use the issue templates

2) remove custom system module

closed 2020-02-20 by SciresM