Atmosphere-NX/Atmosphere #825: Switch crashes when inserting a gamecartridge

Pyjamaa, 2020-02-29:
## Bug Report

### What's the issue you encountered?

When I insert a gamecartridge my Switch(emuMMC) crashes.

### How can the issue be reproduced?

-Boot up the Switch with hekate or fusée into emuMMC
-Insert any gamecartridge

### Crash Report

https://gist.github.com/Pyjamaa/73367b7336c535ce12d2083b92ff1ba9

### System Firmware Version

9.1.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusèe, hekate
- Official release or unofficial build: official 0.10.2
- Do you have additional kips or sysmodules you're loading: N/A
- Homebrew software installed: [ * ]

### Additional context?

I updated my emuMMC from 8.1.0 to 9.1.0 with ChoiDujourNX today. 
On 8.1.0 I had no problems with any cartridges.

Alvonsi, 2020-03-01:
Do you have NOGC Patches enabled?
By default it is in Firmware 9.1.0 enabled, because you will update your gamecard controller and can't use it in any Firmware < 9.1.0 (1.0 - 8.1.0)
Under Hekate Options you can check it.

Pyjamaa, 2020-03-01:
I have it disabled since I'm also using cardridges in Stock(OFW 9.1.0) to play online.
So my gamecard controller should be updated atleast in Stock.

Is there a way to disable the nogc patch in emuMMC?

SciresM, 2020-03-02:
This is a nogc issue -- the error you report is a failure to send the lotus firmware.

Edit your nogc settings in config/bct.ini.

closed 2020-03-02 by SciresM