Atmosphere-NX/Atmosphere #826: Game crashing because of cheat code file inside content folder

isasanti, 2020-02-29:
## Bug Report

### What's the issue you encountered?

After I updated Atmosphere the cheats are crashing the Snack World Dungeon Crawl game. 
I can start the game but when I click to open the save game it crashes and close. 
After I deleted the files with the cheats from the game it works normally (files inside the Atmosphere/content folder). 
How can I use the cheats without the game crashing??

### How can the issue be reproduced?

Every time I put the cheat code files from the **Snack World** game inside the sdcard.

### Crash Report

There is no crash with atmosphere, just crashing the game when I have the folder with the cheats inside the ...Atmosphere/content folder.

### System Firmware Version

9.1.0

### Environment?

- What bootloader: Hekate
- Official release: 1.1.1
- Do you have additional kips or sysmodules you're loading: no
- Homebrew software installed: N/A

### Additional context?

- Additional info about your environment:
I tried updating Edizon and still is crashing the game.
Tried to keep only the folder for the game I need the cheat code.

SciresM, 2020-02-29:
It sounds like whatever cheats you're using are crashing the game, and not like it's an atmosphere issue. Talk to the cheat developers?

isasanti, 2020-02-29:
The cheats were working fine before this change of folders.

isasanti, 2020-02-29:
And even with the game crashing I can see the cheats inside edizon. 
It's only after I hit to continue the saved game/new game that crashes.

isasanti, 2020-03-01:
I just got it to work, I think. xD

Don't really know what made it work but I did two things:
-Copied the folder with the cheats DIRECTLY to Content folder. ( i was creating the titles folder and atmosphere was changing automatically to Content folder.)
-Lunched the game by keeping the L button pressed.

I still think that someone should take a closer look about this crashing thing.....anyway thank you.

closed 2020-03-01 by isasanti

ELY3M, 2020-03-19:
are you aware that holding L button disable the cheats?