Atmosphere-NX/Atmosphere #827: Override language per title doesn't work

hebroin, 2020-03-04:
## Bug Report

### What's the issue you encountered?
Created a config.ini file in the atmosphere\titles\[_titleId_]\ folder with the key _override_config!override_language=fr_ but it didn't do anything.
The game is still in english (system language).

Do I need to activate something for the config.ini file to be used?

### How can the issue be reproduced?
Create a folder "0100CEC003A4A000" in atmosphere\titles
Create a config.ini file with the following line in it _override_config!override_language=fr_
Start the game

--> Still in system language

### Crash Report
N/A

### System Firmware Version
9.1.0

### Environment?
hekate
Atmosphere 0.10.3

### Additional context?
N/A

I'm probably missing a step to active this but didn't find anything about it. 
Tried the discord, but can't get it to log in.
Thanks for the help.

HookedBehemoth, 2020-03-04:
> Atmosphere now uses the /atmosphere/contents directory, instead of /atmosphere/titles.

closed 2020-03-04 by SciresM

hebroin, 2020-03-04:
Thanks for the help.
The file I created was already there (got copied there automagicly I guess) but still doesn't work.

Am I suppose to use a special module for that?
Is there a config to modify in the override_config.ini or system_settings.ini?

SciresM, 2020-03-04:
You need to create an ini file with section override_config, key override_language, value fr.

Look up how to create ini files.

Joeao, 2020-03-18:
`/atmosphere/contents/[titleid]/config.ini` should look like this:
```
[override_config]
override_language = fr
```