Atmosphere-NX/Atmosphere #829: pSNES and pNES controllers aren't responsive anymore.

azoreseuropa, 2020-03-05:
I have a problem with the latest atmosphere release. pNES and pSNES. The controllers aren't responsive! It was worked at 0.10.2. What I did do ? You need to copy system_settings.ini from config_templates in atmosphere and copy it to config folder and edit system_settings.ini using your favourite notepad like this:

enable_deprecated_hid_mitm = u8!0x1
[hbloader]

and save it.
I did that for 0.10.3 and it didnt work anymore.

SciresM, 2020-03-05:
1) reopen with issue templates used if this is a real problem

2) that line needs to be uncommented and in the atmosphere section.

closed 2020-03-05 by SciresM

azoreseuropa, 2020-03-05:
I dont understand. Where is my question going to ?

SciresM, 2020-03-05:
"reopen with issue templates used if this is a real problem"

azoreseuropa, 2020-03-05:
Where ? I never have done this one before.