Atmosphere-NX/Atmosphere #830: Cheats disabled, and depreciated hid do not work when enabled together

piercedfreak, 2020-03-05:
## Bug Report

</br>

### What's the issue you encountered?

[ I loaded all the cheats from Edizon onto my sd card, then set dmnt_cheats_enabled_by_default = u8!0x0, I already had enable_deprecated_hid_mitm = u8!0x1 set. Tried to boot, and sits at a black screen after the Nintendo logo. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ I loaded all the cheats from Edizon onto my sd card, then set dmnt_cheats_enabled_by_default = u8!0x0, I already had enable_deprecated_hid_mitm = u8!0x1 set.]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

9.1.0

### Environment?

-  hekate
- Official release or unofficial build: 
  - [ 0.10.2 offical build]
- Do you have additional kips or sysmodules you're loading: Emuiibo
- Homebrew software installed: [ ]

### Additional context?

- Additional info about your environment:
- [  ]

SciresM, 2020-03-05:
Does the error occur with emuiibo removed?

piercedfreak, 2020-03-05:
I did not do any further testing after commenting out enable_deprecated_hid_mitm = u8!0x1, I will do some more testing once I get home this evening to see if removing emuiibo has any effect on it.

SciresM, 2020-03-05:
fwiw my console boots fine with cheats disabled + deprecated hid mitm turned on, so I can't reproduce.

piercedfreak, 2020-03-07:
Now I can not get it to reproduce. I fought with it for over an hour before. Guess it was a glitch.

closed 2020-03-07 by piercedfreak