Atmosphere-NX/Atmosphere #832: fatal error occured when running atmosphere (titleid 0100000000000008)

TheRealDarklord, 2020-03-08:
## Bug Report

### What's the issue you encountered?

I just updated to atmosphere 0.10.3 and booted my switch and I get the following error:

```
a fatal error occured when running atmosphere
title id: 0100000000000008
error desc: std::abort() called (0xffe)
report saved to /atmosphere/fatal_errors/report_00000000075634fd.bin
```

### How can the issue be reproduced?

I just boot in RCM Hekate, select the Atmosphere CFw to boot and wait. After all of the atmosphere images were shown the above error pops up. Every time I try to boot since the update to 0.10.3.

### Crash Report

The content of the fatal_error file (as it is created as named in the error message) looks like a binary format. I'm not sure how to paste this into a gist. Is there any tool to read the contents of this bin-file in a human readable way? Or should I upload the bin-file anywhere else?

### System Firmware Version

9.1.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: hekate
- Official release or unofficial build: official release 0.10.3
- Do you have additional kips or sysmodules you're loading: just the ones needed for fusee-primary as stated in hekate repo
- Homebrew software installed: CheckPoint, EdiZon, nxmtp, nx-shell,

### Additional context?

- Additional info about your environment: N/A

TheRealDarklord, 2020-03-08:
sry to bother you, I forgot to copy the /sept folder to the sdcard...  
it's now working, my bad...

closed 2020-03-08 by TheRealDarklord

Xyphoseos, 2020-03-18:
I've got it too, but all folder are present...