Atmosphere-NX/Atmosphere #835: error: 'setsysBindFatalDirtyFlagEvent' was not declared in this scope

Kronos2308, 2020-03-11:
## Bug Report

I am trying to compile atmophere but I always get the same error
 (error: 'setsysBindFatalDirtyFlagEvent' was not declared in this scope)
  at one point in the compilation
I am using the binary files of the sept of the release but what it does not compile is the fatal

### What's the issue you encountered?

```make
In file included from c:\users\administrador\documents\github\atmosphere\libraries\libvapours\include\vapours\results.hpp:23,
                 from c:\users\administrador\documents\github\atmosphere\libraries\libvapours\include\vapours.hpp:25,
                 from c:\users\administrador\documents\github\atmosphere\libraries\libstratosphere\include\stratosphere.hpp:20,
                 from C:/Users/Administrador/Documents/GitHub/Atmosphere/stratosphere/fatal/source/fatal_config.hpp:17,
                 from C:/Users/Administrador/Documents/GitHub/Atmosphere/stratosphere/fatal/source/fatal_config.cpp:16:
C:/Users/Administrador/Documents/GitHub/Atmosphere/stratosphere/fatal/source/fatal_config.cpp: In function 'Handle ams::fatal::srv::{anonymous}::GetFatalDirtyEventReadableHandle()':
C:/Users/Administrador/Documents/GitHub/Atmosphere/stratosphere/fatal/source/fatal_config.cpp:28:28: error: 'setsysBindFatalDirtyFlagEvent' was not declared in this scope; did you mean 'setsysGetFatalDirtyFlags'?
   28 |             R_ABORT_UNLESS(setsysBindFatalDirtyFlagEvent(&evt));
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
c:\users\administrador\documents\github\atmosphere\libraries\libvapours\include\vapours\results\results_common.hpp:253:39: note: in definition of macro 'R_ABORT_UNLESS'
  253 |         const auto _tmp_r_abort_rc = (res_expr); \
      |                                       ^~~~~~~~
make[3]: *** [/opt/devkitpro/devkitA64/base_rules:14: fatal_config.o] Error 1
make[3]: *** Waiting for unfinished jobs....
```

### How can the issue be reproduced?

[Just compile]

### Crash Report

[ ]</br>

### System Firmware Version

[]

### Environment?

[Devkitpro windows 10]

### Additional context?

 i update manualy the lib nx from the source

shchmue, 2020-03-11:
that had the wrong name in libnx, Atmosphere will have to be updated to use the correct name: `setsysAcquireFatalDirtyFlagEventHandle`

Kronos2308, 2020-03-11:
for the moment i change
setsysBindFatalDirtyFlagEvent
for
setsysAcquireFatalDirtyFlagEventHandle

and should compile ?

shchmue, 2020-03-11:
yes

Kronos2308, 2020-03-11:
ok thaks

closed 2020-03-11 by SciresM