Atmosphere-NX/Atmosphere #837: Error Code: 2162-0002 (0x4a2)

xmagcx, 2020-03-18:
## Bug Report

Hi guys waiting to be fine I have the error 010000000000100f (attached log of the folder fatal report). As evidence it happens to me with the game super smash bros ultimate when starting both story mode or any other mode. What I have tried so far:
1- Just play the version without updates.
2- install the latest update
3- install different updates and nothing

Please I need your help I have tried and tried everything in my power

[01584409357_010000000000100f.log](https://github.com/Atmosphere-NX/Atmosphere/files/4346546/01584409357_010000000000100f.log)
[image]

SciresM, 2020-03-18:
Please re-open using the issue templates.

closed 2020-03-18 by SciresM

xmagcx, 2020-03-18:
What's the issue you encountered?

Hi guys waiting to be fine I have the error 010000000000100f (attached log of the folder fatal report). As evidence it happens to me with the game super smash bros ultimate when starting both story mode or any other mode. What I have tried so far:
1- Just play the version without updates (Smash Bross Ultimate Game).
2- install the latest update (Smash Bross Ultimate Game)
3- install different updates and nothing (Smash Bross Ultimate Game)

How can the issue be reproduced?
 

Crash Report
 
https://gist.github.com/xmagcx/3478a5e131d27b195fee164638cd1555

System Firmware Version
9.2.0
 
Environment
bootloader Atmosphère launched by: Atmosphère
Official release or unofficial build: N/A
 Official release version 0.10.4-master-e8dfe047 
 If using an unofficial build, include details on where/how you acquired the build. 
 Ex: N/A
 Ex: N/A
Do you have additional kips or sysmodules you're loading: no
Homebrew software installed:  *  atmosphere
Additional context: N/A
Additional info about your environment: N/A
 Any other information relevant to your issue: Attached image for reference error
[image]