Atmosphere-NX/Atmosphere #841: Bypass internet access check when connecting to a WiFi (or eth) network (if possible)

kapeas, 2020-03-19:
## Feature Request
</br>

### What feature are you suggesting?
#### Overview:
- Bypass the internet connectivity check when conecting to a WiFi network</br>

#### Smaller Details: 
- I think it may be useful to be able to connect to a Wifi or ethernet network with no internet access. This will allow to leverage some lan features for homebrew reducing the risk to get banned. For example, you can create a WiFi access point using an old android phone without sim and no 3/4/5g data, and then connect your Switch and PC to that network so both can communicate via wifi without risking sending data to N. Since there's no internet, switch can't send anything outside lan.</br>

#### Nature of Request:  
- I would like that would be possible to use homebrew and take advantage of the wifi features like ftp, streaming and other cool stuff even if the wifi has no internet access, or it is capped, avoiding the ban risk.
- Addition Bypass internet connectivity check
- Change Internet connectivity check (to not check, or ignore results)
- Removal Internet connectivity check

### What component do you feel this would best fit within?
- [Fusée] Unknown</br>
- [Exosphère]Maybe here</br>
- [Stratosphère]Or here</br>

- Custom system modules maybe?</br>
- [Thermosphère] N/A</br>
- [Troposphère]Unknown</br>

### Why would this feature be useful?
Avoid the personal/hardware information sending to Ns servers, keeping your stuff private and ban safe (presumably)

If possible, and useful for more people, would love this feature implemented on atmosphere.
Thanks everyone contributing for your efforts.

spling-du-futur, 2020-06-06:
connection test don't send telemetry
it just try to ping nintendo servers

Slluxx, 2020-06-15:
+1 for that feature. id be also glad to have that.

spling-du-futur, 2020-06-16:
yes like a "fake-internet" function
i think you can patch HOS for this

kapeas, 2020-06-17:
> connection test don't send telemetry
> it just try to ping nintendo servers

Hi, the point of the feature request is to be able to connect to a network **with no internet**, that won't be able to resolve DNS names, nor connect to any external site. It does not matter to me if it only pings N servers, since there won't be any valid external ip, nor dns server to resolve the external request (ping to N servers). Feature request is to be able to connect to a network even if those ping fail, as they should if no internet output on the network.

Grass0001, 2020-07-08:
would also like to have this feature, mostly just out of convenience

JrRadiant, 2020-10-27:
Nice idea, btw!

kapeas, 2021-02-02:
This feature requests seems unnecesary with the new versions of atmosphere including the DNS mitm sysmodule.

Thanks A LOT for adding this sysmodule. Thank you everyone involved in atmosphere development and this particular sysmodule.

closed 2021-02-02 by kapeas