Atmosphere-NX/Atmosphere #843: Switch Doesnt boot past switch logo Error Code: 2162-0002 (0x4a2)

AkiraYuki21, 2020-03-20:
## Bug Report

Error Code: 2162-0002 (0x4a2)

### What's the issue you encountered?

Can't get past Nintendo Switch Logo regardless if OFW or CFW. Always get this error:
Error Code: 2162-0002 (0x4a2)

### How can the issue be reproduced?

N/A

### Crash Report

https://gist.github.com/AkiraYuki21/62bc111be31317c462f0f4db5ed8eb04

### System Firmware Version

9.1.0

### Environment?

fusee-primary.bin

### Additional context?

N/A

nwhiteman42, 2020-03-20:
If this happens on OFW as well this is not Atmospheres issue and this should be closed.  Check out these links for help. 
 https://www.reddit.com/r/NintendoSwitch/comments/5xidmr/there_is_a_system_breaking_crash_error_code/
https://en-americas-support.nintendo.com/app/social/questions/detail/qid/72362/~/possible-fix-for-error-code%3A-2162-0002/comment/21159
https://en-americas-support.nintendo.com/app/answers/detail/a_id/22596/~/error-code%3A-2162-0002

It might be a hardware issue with the wifi chipset

SciresM, 2020-03-20:
crash in the usb sysmodule really wouldn't be atmosphere-related, since we don't touch anything that impacts usb. If it happens on OFW, I'm gonna say it's not an Atmosphere issue, and it's probably a hardware problem.

closed 2020-03-20 by SciresM