Atmosphere-NX/Atmosphere #844: Animal Crossing game crashing in atmosphere after the game tries to save for first time.

ELY3M, 2020-03-21:
## Bug Report
Animal Crossing on game cart crashing on the first save.  

I have 3 switches with atmosphere running.  
2 have emunands and both are online.  
1 is banned and atmosphere is running over its sysnand.   

I been playing this game for hours. 
I started over many times.  
the game kept crashing when it is first saving the game.  

I decided to reboot one of my switches to normal stock sysnand.  
the game did saved fine without crashing.   

I wonder why....  I wonder if game dev added in "anti-atmosphere" code in the game.   
I would hate that company so much and return my game cart.    

### What's the issue you encountered?
Describe the issue in detail and what you were doing beforehand
I went to Target and bought the game cart.  
I played it for many hours - started over and crashing at first save.... 

Did you make any changes related to Atmosphère itself?
No changes.  

### How can the issue be reproduced?
1. put in game cart of Animal Crossing.  
2. let it update if it got update.   
3. be sure you are online and running in atmosphere.  
4. play the game from the beginning...  
5. continue play... 
6. when Tim Nook start to say about game saving stuff....  
7. it will auto save and CRASH....  software closed...  

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
all crash logs from all of 3 switches.     
https://www.dropbox.com/s/xb69r9qk65letx4/!bug report - AC crashes.zip?dl=1

### System Firmware Version
9.2.0  

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:  Fusee-primary.bin 
- Official release or unofficial build:  Official build  0.10.5 

- Do you have additional kips or sysmodules you're loading:
sys-ftpd-light  tesla overlay  

- Homebrew software installed: [ * ]
ftpd
edizon
nxthemes
payload launcher 
capturesight 
rcmreboot
few of my own homebrew apps

fincs, 2020-03-21:
Remove custom sysmodules and third party software. Use completely [vanilla and unmodified Atmosphère](https://github.com/Atmosphere-NX/Atmosphere/releases/latest).

I run AC with Atmosphère without issues.

ELY3M, 2020-03-21:
I am going try without any sysmodules.

ELY3M, 2020-03-21:
I am still crashing without sysmodules and many of homebrew apps.  I am on vanilla and not modified atmosphere.   I will upload more crash logs.

ELY3M, 2020-03-21:
I also tested atmosphere over sysnand and the game still crashed even it was saved in normal sysnand.

ELY3M, 2020-03-21:
here are crash logs from testing on cfw over sysnand   
https://www.dropbox.com/s/vqzcjp5ahgvvjt7/switch3-sysnand.zip?dl=1

SciresM, 2020-03-21:
Okay, let's test this properly. I have been playing animal crossing on sysnand pretty much all day, so I'm reasonably confident atmosphere's fine.

Please test the following:
-Rename your atmosphere folder to atmosphere_old.
-Extract 0.10.5 release zip to SD card, using stock atmosphere configuration with no changes.
-Test on sysnand booting using fusee-primary with no other changes.

ELY3M, 2020-03-22:
I finally know why my game been crashing.   I was able to save the game in both sysnand and emummc.  then I reinstalled my themes.  guess what my game crashing on loading saves.   
It may be my custom fonts?

ELY3M, 2020-03-22:
It was my custom fonts that is causing game to crash on saving or loading.

SciresM, 2020-03-22:
Yeah. That's why I had you test with actually clean atmosphere. Not really a bug -- your font causes the game to crash. Probably it's expecting a character you don't provide? Anyway, closing the issue.

closed 2020-03-22 by SciresM