Atmosphere-NX/Atmosphere #846: Blank Screen ....again

chubbysocks, 2020-03-21:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

### System Firmware Version

X.X.X</br>
[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: 
  - [ Official release version x.x.x (or) unofficial build ]
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading:
- Homebrew software installed: [ * ]

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

chubbysocks, 2020-03-21:
I'm was on 8.1.0/0.9.4 atmos. I tried to update the firmware to 9.2.0 with ChoiDujourNX. When I went to reboot I injected the payload with my RCMloader and a little message popped up and in the right hand corner, and now just blank screen. 

I'm thinking I should've updated atmos first before the switch firmware?? Anyhow can you good people help me out? You fixed my problem last time, cheers guys.

SciresM, 2020-03-22:
1) actually fill out the issue templates

2) ...just update atmosphere?

closed 2020-03-22 by SciresM

chubbysocks, 2020-03-22:
> 1. actually fill out the issue templates
> 2. ...just update atmosphere?

I've updated atmosphere, nothing happens, just a blank screen

SciresM, 2020-03-23:
@chubbysocks "updated atmosphere" doubt it. You're probably pushing an older bootloader or something instead of updated fusee-primary/hekate.

chubbysocks, 2020-03-23:
> @chubbysocks "updated atmosphere" doubt it. You're probably pushing an older bootloader or something instead of updated fusee-primary/hekate.

Updated the payload, extracted the newest version of atlasnx kosmos, and I've updated to atmos 0.10.5.