Atmosphere-NX/Atmosphere #847: A fatal error occurred when running Atmosphere

HyImTIm, 2020-03-23:
## Bug Report

A fatal error occurred when running Atmosphere.
Title ID: 010041544d530000
Error Desc: std::abort() called (0xffe)
Report saved to /atmosphere/fatal_errors/report_0000000061b2685.bin

### What's the issue you encountered?

After booting up Atmosphere, through Hekate, screen goes black and I receive the fatal error message

### How can the issue be reproduced?

Boot to RCM
Launch Hekate
Boot Atmosphere through either fusèe-primary or emunand

### Crash Report

https://github.com/HyImTIm/bugs/blob/master/report_00000000061b2685.bin

### System Firmware Version

9.2.0

SciresM, 2020-03-23:
Run archive bit fixer in hekate, you have invalid archive bits (probably you use macOS) that are causing atmosphere to be unable to interact with the SD card the way it needs to.

closed 2020-03-23 by SciresM

HyImTIm, 2020-03-23:
I am using windows and when I ran it, it still persists

SciresM, 2020-03-23:
If using windows, try deleting atmosphere folder on SD and re-extracting from zip. Your crash report indicates that atmosphere was unable to open the backup path for your PRODINFO -- this can only happen when the archive bit is set on the ams folder or a subfolder.

HyImTIm, 2020-03-23:
Okay so I am getting two different errors. When I boot atmosphere from launch<CFW (EMUMMC), I get an error saying 

Wrong ini cfg or missing files
Failed to launch HOS

But when I launch though payload<fusee-primary.bin I get the same fatal error message as I stated earlier

Also I tried what you suggested and nothing still

HyImTIm, 2020-03-23:
Update

I still cannot boot through launch<CFW (EMUMMC). I get the same error message

However I am able to boot through payload<fusee-priamry.bin. The issue is that when I access the hbmenu, there are no applications and it crashes when I tap on the screen. The error code is 2168-0002 (0x4a8)

CTCaer, 2020-03-23:
```
Wrong ini cfg or missing files
Failed to launch HOS
```
Irrelevant with your issue. This means that your hekate boot entry is wrong or you are missing files.

As for the archive bit, you need to run both options in hekate, which you didn't run.

Reminder that Windows **also** sets that bit on new or changed files. And mostly depends on the program that created or changed that file. (So either you didn't delete the atmosphere folder when extracting the files anew or your extractor sets the bit)

HyImTIm, 2020-03-23:
My apologies but I am not sure what you mean by all that lol

HyImTIm, 2020-03-23:
Okay I got it all working now thank you.