Atmosphere-NX/Atmosphere #848: Error Code: 2162-0002 (0x4a2)

sttba, 2020-03-23:
## Bug Report

Error reported by xmagcx, and I have the same result, error 010000000000100f (attached log of the folder fatal report). 

### What's the issue you encountered?

error 010000000000100f.  

### How can the issue be reproduced?

As evidence it happens to me with the game super smash bros ultimate when starting both story mode or any other mode and mario oddisei menu.

1. Just play the version without updates (Smash Bross Ultimate Game).
2. install the latest update (Smash Bross Ultimate Game)
3. install different updates and nothing (Smash Bross Ultimate Game)
4. Start game in story mode, after the intro video, system crashes 

odissei
1. Open Action Guide menu
2. Open Bonus Travel hints menu
3. Exit previous menu

### Crash Report

https://gist.github.com/xmagcx/3478a5e131d27b195fee164638cd1555

### System Firmware Version

9.2.0

### Environment?

bootloader Atmosphère launched by: Atmosphère
Official release or unofficial build: N/A
Official release version 0.10.4-master-e8dfe047
If using an unofficial build, include details on where/how you acquired the build.
Ex: N/A
Ex: N/A
Do you have additional kips or sysmodules you're loading: no
Homebrew software installed: * atmosphere
Additional context: N/A
Additional info about your environment: N/A

### Additional context?

https://user-images.githubusercontent.com/20447334/76958959-65104f80-68f7-11ea-8c84-618e25f7ac8c.jpg

xmagcx, 2020-03-24:
Hello @SciresM, you could review this case, I think it affects many people,

regards

xmagcx, 2020-03-25:
Hello @SciresM, yesterday I played Mario Odissey and had the same error when advancing to the second stage,

regards

aminxsaidoun, 2020-03-26:
Hi @SciresM , apologies to bother you with this under our stressful times with COVIG-19. But I am having the same issue with loads of different people under the same boat. 

Will you be able to address the issue, I'm stuck and my wife really wants to play with the system! :(

Thanks for your hard work,
PandanV

Alvonsi, 2020-03-29:
Hello Guys,
I have both games and can't reproduce anything...
I'm using minimal install of atmosphere! 0.9.2 | 0.10.5 E 
With SigPatches and one small sysmodule.
And nothing else.
Games in Version 1.0.0
I think u have anything installed or different...
Are u using nxthemes or something?
best regards

fincs, 2020-03-29:
> SigPatches

Piracy is not supported.

> one small sysmodule.

Custom sysmodules and themes should be removed before seeking support here, in order to confirm that it really is an issue with Atmosphère.

xesxen, 2020-04-01:
Hey all,

I've got the same error when I undocked my 1.0.0 switch. This one has only seen one game (which wasn't booted) and is running a clean Atmosphere installation.

Crash log: https://gist.github.com/xesxen/b9078408437d80655bf67269f84481b8

xmagcx, 2020-04-01:
Guys the only solution since these developers will not catch them, is to invest a little money and buy sx-SO, I bought it and no problem with any game, it is also very easy to use,

Regards

Juanperezc, 2020-04-09:
I have the same problem with mario od and super smash. apparently I'm not the only one

TheZckboy, 2020-04-15:
does someone have tried the newest version of atmosphere (0.11.1) with the new ofw(10.0.0)? is this issue still there?

sttba, 2020-04-17:
I updated to 10.0.0-0.11.1 and the error was gone, I had to delete the atmosphere folder and replace it with the new version, I think is possible, not sure, that a clean installation fix this issue (suggested before) in case you don't want to upgrade.

closed 2020-04-22 by SciresM

Alxzu, 2021-02-01:
Do you think this may happen due to having smash bros DLCs that require 11.0.1? I'm getting the same issue with 11.0.0

ELY3M, 2021-02-01:
you need to create new issue and follow the template!    .