Atmosphere-NX/Atmosphere #850: Atmosphere crashes after the switch logo when loading emunand

Kosuke-ak, 2020-03-24:
Bug Report
[ Note: If the bug or crash you encountered is related to; ]

[ - software used to make "backups", ]

[ - software explicitly distributed for piracy, etc ]

[ then contributors will not provide support for your issue and your issue will be closed. ]

What's the issue you encountered?
[ Describe the issue in detail and what you were doing beforehand. ]
try to update emunand 4.1.0 to 8.1.0 with choix du jour but after rebooting hetake, when emunand is booting, just after switch logo, it crash.
[ Did you make any changes related to Atmosphère itself? ]
NO
[ If so, make sure to include details relating to what exactly you changed. ]

How can the issue be reproduced?
[ * ]

[ Include a detailed step by step process for recreating your issue. ]

Crash Report
[ Crash reports can be found under /atmosphere/crash_reports. ]

[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a gist and pasting the link here. ]

[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]

System Firmware Version
4.1.0

Environment?
What bootloader (fusèe, hekate, etc) was Atmosphère launched by:

Official release or unofficial build: Official Hetake 5.1.2

Do you have additional kips or sysmodules you're loading: no

Homebrew software installed: [ none ]

Additional context?
Additional info about your environment:
[ Any other information relevant to your issue. ]
I tested with factory reset option in choix du jour, all be done but 8.1.0 want to erase the sd card (it don't leave me the choise) and after doing it, the switch reboot and after rebooting hetake, when emunand is booting, just after switch logo, it crash.

Sorry for not using the template at the first time.

SciresM, 2020-03-24:
You're using custom sysmodules of some kind with high certainty -- and exceeding the session limit for some service. Remove the custom sysmodules.

Kosuke-ak, 2020-03-24:
all done
thanks

closed 2020-03-24 by SciresM