Atmosphere-NX/Atmosphere #851: Input Controllers on Emulators

limrrr, 2020-03-24:
### What's the issue you encountered
Updated from Atmosphere 9.2 to 10.4, and afterwards, emulators such as PSNES and VBA-next does not respond to input controls.

### Crash Report
Atmosphere did not crash. 

### System Firmware Version

9.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee
- Official release or unofficial build: 
  - Official release version 9.2.0 or Atmosphere 10.4
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: Atmosphere 10.4

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-03-24:
Enable the deprecated hid mitm or update your Homebrew.

closed 2020-03-24 by SciresM

limrrr, 2020-03-24:
Thanks for the quick response. What if the problem persists if the deprecated hid mitm is already enabled in the system_settings?