Atmosphere-NX/Atmosphere #852: A game that crash when loading layeredfs

sawada3014, 2020-03-25:
error report↓
https://mega.nz/#!j6pxiAYA!sd9KZ2X8-58dlDssNtBTd24s2Mh204873CGWyMytzzU

error message↓
panic occuerd while runnning atmosphere.
title id 01004154453000

Crashed a cfw(9.2.0) when I try to launch "animal crossing new horizons"
I used layeredfs then.
I guess thet this crash came from trying to load layeredfs mods(romfs).
Because This problem was fixed by deleting dir (contents / 01006F8002326000 / romfs / ...).
But, at the other game , such probrem never occuerd.
(Because this game only supports version 9.1.0 or later?)
I report this problem because at description of latest release I found the sentence  such as
 "If you encounter a game that exhausts ams.mitm's memory (crashing it) when loading layeredfs mods, please contact SciresM."

I want to load layeredfs mods on this game,Please fix this probrem
And I'm sorry for my poor English.

deadair3210, 2020-03-25:
remake the issue with the template

SciresM, 2020-03-25:
Update atmosphere.

closed 2020-03-25 by SciresM