Atmosphere-NX/Atmosphere #855: No main boot entries found...

aminxsaidoun, 2020-03-26:
I began playing Pokemon Sword earlier today, after a few tries I kept getting a crash screen saying to press the Vol+ to reboot...etc

Worked a few times, afterwards, I got a black screen saying to reboot by pressing power button. unfortunately, that worked for a short while and then it would crash every 2mins.

Now when I boot up the system and select the Launch button from Hekate I constantly get this page... No main boot entries found...
You can use the following entry to boot stock. or use More configs button for more boot entries.

Please help thank you so much!

deadair3210, 2020-03-26:
going to assume that your SD card is either dying or corrupted

PandanV, 2020-03-26:
Thanks for the response, made me smile to see it. Under these strange times
we live in now.

Good to see the community is alive. So probably should either format it
clean again, or in other words replace it completely right?

Appreciate your help

SciresM, 2020-03-26:
Please re-open using the issue templates, which ask you to provide important diagnostic information.

In addition, I don't maintain hekate. I agree it's probably an sd card issue.

closed 2020-03-26 by SciresM

PandanV, 2020-03-26:
Apologies very new to all of this homebrew community

Really appreciate your support thank you.