Atmosphere-NX/Atmosphere #861: Software wont boot up after update.

lajax01, 2020-03-30:
## Bug Report

[ If any section does not apply, replace its contents with "N/A". ]</br>
[ Lines between [ ] (square brackets) should be removed before posting. ]</br>

[ * ]</br>
[ Note: If the bug or crash you encountered is related to; ]</br>
[ - software used to make "backups", ]</br>
[ - software explicitly distributed for piracy, etc ]</br>
[ then contributors will not provide support for your issue and your issue will be closed. ]</br>

### What's the issue you encountered?
Software wont boot up, that is worked just fine previous fw version (9.1.0)
Error log attached.

[ Describe the issue in detail and what you were doing beforehand. ]</br>
[ Did you make any changes related to Atmosphère itself? ]</br>
[ If so, make sure to include details relating to what exactly you changed. ]</br>

### How can the issue be reproduced?
Start same software, but anything else works just fine.

[ * ]</br>
[ Include a detailed step by step process for recreating your issue. ]</br>

### Crash Report

[ Crash reports can be found under ``/atmosphere/crash_reports``. ]</br>
[ If your issue caused Atmosphère to crash, include the crash report(s) by creating a [gist](https://gist.github.com/) and pasting the link here. ]</br>
[ If you don't include a crash report in instances of crash related issues, we will ask you one to provide one. ]</br>

9.2.1 System Firmware Version 

9.2.0</br>
[01585601148_010075d00e8ba000.log](https://github.com/Atmosphere-NX/Atmosphere/files/4405396/01585601148_010075d00e8ba000.log)

[ Replace X's with system firmware version at time of crash. ]</br>
[ You can find your firmware version in the Settings -> System, under "System Update". ]</br>
[ If it says "Update Pending", you can clear the pending update by rebooting to Maintenance Mode. ]</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by:
- Official release or unofficial build: Official
  - [ Official release version x.x.x (or) unofficial build ] 0.10.5
  - [ If using an unofficial build, include details on where/how you acquired the build. ] 
     - [ Ex: Self-compilation ]
     - [ Ex: Kosmos' distribution of Atmosphère ]
- Do you have additional kips or sysmodules you're loading: Yes.
- Homebrew software installed: [ * ] Lots of.

### Additional context?

- Additional info about your environment:
- [ Any other information relevant to your issue. ]

SciresM, 2020-03-30:
9.2.0 and 9.1.0 have no difference in atmosphere. The single line change is applied in both firmware versions under ams.

Does issue occur without custom sysmodules/homebrew?

lajax01, 2020-03-31:
Yes. Usualy exefs paches, and content paches was applyed on this title, but the crash on boot occured without them either. Tesla running newly in the background, with custom overlays like sysclk, and edizone.
Could this the problem?

SciresM, 2020-03-31:
Tesla is a custom sysmodule. I asked you to remove custom sysmodules and test without them, please.

closed 2020-03-31 by SciresM

lajax01, 2020-03-31:
Ok, will do. Ill back in 12 hrs with the result. Thanks for your time, and cooperation! Muck appreciate!