Atmosphere-NX/Atmosphere #864: Blank screen while booting into emuMMC (after a reboot from emuMMC)

hazzzy, 2020-04-02:
## Bug Report

Hi, I really have no idea where else I'd report this first so I'll try my luck here.
This issue is sort of weird and it only occasionally happens (although almost guaranteed lately), most often after rebooting from CFW into hekate and then trying to boot into CFW again.

### What's the issue you encountered?

Switch gets stuck while booting emuMMC after the "Nintendo Switch" logo **fades** out.
The screen is blank and the backlight is turned **ON**. 
The Switch does not respond to any button presses or combinations and needs to be force-shutdown by holding the power button for ~10 seconds.
If left untouched for some time while it is stuck, the backlight does turn OFF, but turns ON again immediately after pressing any button or touching the screen.

Additionally, I've sometimes had an issue where after force-shutdown and reboot, the switch gets back into RCM after the sept logo.

### How can the issue be reproduced?

1. Boot into emuMMC via hekate
2. Successfully boot into atmos
3. Reboot through the ``restart`` option in power menu (holding power button for ~3 seconds)
    (I've noticed that this also happens if I shutdown and inject hekate again manually)
4. Successfully reboot into hekate
5. Launch emuMMC
6. Get stuck at blank screen after Nintendo Switch logo fades out

### Crash Report

``crash_reports`` and ``fatal_errors`` folders are empty/don't exist

### System Firmware Version

9.2.0

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: **hekate 5.1.3**
- Official release or unofficial build: **Official release 0.10.5**
- Do you have additional kips or sysmodules you're loading: **No**
- Homebrew software installed: Not sure if it's relevant, but just the typical stuff such as:
    - appstore, awoo installer, checkpoint, choidujournx, edizon, ftpd, goldleaf, kosmostoolbox (it's a leftover), ldnmitm_config (also leftover), lockpick, nxdumptool, nx-shell, nxthemesinstaller, sigpatch-updater (never used), switch_90dns_tester

### Additional context?

**Additional info about your environment:**

This is a newly hacked switch and this has been happening for a while, although I can't really remember for how long, could be from the very start.
The firmware that was installed before the hacking was 9.1.0. After setting up atmos (or kosmos), I've updated the firmware to 9.2.0 with ChoidujornX on both emuMMC and stock sys. I haven't had any issues with stock sys.

The sdcard used is Toshiba Exceria 128gb. I've tested it multiple times on both Windows(with h2w) and linux (with f3) and all the times there were 0 errors or issues.
I've done the emuMMC setup in 3 different ways just to rule out that as well:
1. On Windows using DiskGenius - [guide](https://switch.homebrew.guide/emummc/windows)
2. Using TegraExplorer's SD format option for emuMMC - [guide](https://nh-server.github.io/switch-guide/user_guide/emummc/partitioning_sd/)
3. On Linux using gparted and overwriting the emuMMC partition beginning and end - [guide](https://switch.homebrew.guide/emummc/linux)

At first my setup was the latest Kosmos. After experiencing these issues I've tinkered with Kosmos and tried different things from using sigpatches and different booloader configs to removing modules and homebrew, but to no avail.
After that I've done a manual atmos setup with hekate + kosmos sigpatches + some sys modules like sys_con. This also produced the same behaviour, so I removed the sys_modules. It seemed as it would work for a while with no issues (maybe for a week) before I randomly started experiencing issues again (I've not modified anything regarding atmos or hekate)

My latest setup which I'm basing this report on is the following:

1. completely wiped sdcard
2. sdcard setup for emummc using [this guide](https://switch.homebrew.guide/emummc/linux)
3. unpacked atmosphere 0.10.5 to sdcard (except switch folder)
4. unpacked hekate 5.1.3 to sdcard
5. replaced ``atmosphere/reboot_payload.bin`` with hekate's payload ``hekate_ctcaer_5.1.3.bin`` (renamed to reboot_payload.bin)
6. add this bootloader config: [hekate_ipl.ini](https://nh-server.github.io/switch-guide/files/emu/hekate_ipl.ini)
7. copied over the ``switch``folder containing previously listed homebrew
8. emuMMC created and enabled using hekate

I've also tried using hekate's "unset archiving bit" option but it made no difference to the problem.

AutoRCM is turned **ON**

I guess this could also be a hekate issue, sorry if it's not related to atmos

themrheng99, 2020-04-04:
facing same problem here

lydiagreenn, 2020-04-07:
I have the same problem as you. When I formatted the tf card, it was solved.

ghz2020, 2020-04-09:
I’m having this same issue any way to solve it formatted sd card problem still occur

themrheng99, 2020-04-09:
i solve the problems using this tutorial 
Cleanup the NAND first -
https://www.youtube.com/watch?v=gVjF_MKjSv8
Setup EMUMMC - 
https://www.youtube.com/watch?v=YW5OxN4SzUQ
use the files give in the description of video
work perfectly now

therealprocyon, 2020-04-09:
I personally would recommend against using video guides, as they get outdated rather quickly.

fincs, 2020-04-09:
> awoo installer, (...) goldleaf, (...) sigpatch-updater

Piracy is not supported.

closed 2020-04-09 by fincs

hazzzy, 2020-04-09:
> 
> 
> > awoo installer, (...) goldleaf, (...) sigpatch-updater
> > (...) kosmos sigpatches (...)
> 
> Piracy is not supported.

I suggest you at least attempt to read the entire report. The setup that this report was based on was stock atmos+hekate with NO homebrew used or being present on the SDcard, and the error occurs before even booting into the CFW.

It's like I'm on the 2000s forums all over again.