Atmosphere-NX/Atmosphere #868: OFW boot corrupted, really need help!!!

zzz6839, 2020-04-10:
Normal booting to OFW no response without injector(when I press the power button it shows black screen), I highly suspect the whole booting system is corrupted, RCM works fine, I can run into injected boot system(with USB dongle). but cannot boot into EMMC(cannot boot into the stock system now), I searched every existing similar problem on the internet, reformat SD card to Fat32 now, even tried different CFW, disable autoRCM and auto Boot, but nothing works when you try to boot into stock and EMMC system.

What to do..........
Rip my Switch

zzz6839, 2020-04-10:
Only RCM works now, booting to stock or CFW would remain black screen after the NS-atmosphere logo(I tried Kosmos but also black screen after Kosmos logo appeared).

zzz6839, 2020-04-10:
this whole thing happened when I tried to update Hbg shop. it froze when I press B to apply for the update, after waiting for a long time I decided to force shutdown my switch, and since then I've never be able to boot into the system again..... and yes I was using Exfat before.......

bakugo, 2020-04-10:
Restore a backup, problem solved

zzz6839, 2020-04-10:
> Restore a backup, problem solved

I have no backup, would someone send me your backup to me pls orz

zzz6839, 2020-04-10:
> > Restore a backup, problem solved
> 
> I have no backup, would someone send me your backup to me pls orz

I know im playing with fire.......im sorry......

deadair3210, 2020-04-10:
1. No Piracy Tools
2. Backups are console-specifc. Even if someone did send you one, it would only make things worse

closed 2020-04-10 by zzz6839