Atmosphere-NX/Atmosphere #869: Fail to boot to emuMMC after reboot

brnor, 2020-04-10:
## Bug Report

### What's the issue you encountered?

Booting to emuMMC fails - Switch either gets stuck at a blank screen or reboots to RCM (or it shutdowns, not clear).

Booting to sysMMC succeeds without any issues every time (tried up to 10 consecutive reboots). This was tested by setting `enabled=0` in emummc.ini

### How can the issue be reproduced?

0. Have emuMMC enabled
1. Inject fusee-primary 
2. One of the two happens (usually in this order):
    a. Boot process passes the "Nintendo Switch" logo and gets stuck at a blank screen (black but 
       with backlight enabled)
    b. Boot process passes the Atmosphere splash art and gets 'stuck' at a completely black screen - 
        pressing the power button once makes the Switch enter RCM (I'm guessing this is an 
        attempted shutdown then?)

### Crash Report

There are no generated logs which I could find.

### System Firmware Version

sysMMC - 9.2.0</br>
emuMMC - 9.2.0</br>

### Environment?

- What bootloader (fusèe, hekate, etc) was Atmosphère launched by: fusee-primary.bin
- Official release or unofficial build: Official release version 0.10.5
- Do you have additional kips or sysmodules you're loading: No
- Homebrew software installed: reboot-to-payload.nro

### Additional context?

- first sdcard partition is FAT32
- emuMMC using a partition, sdcard setup made using TegraExplorer 
- emuMMC config: [link](https://paste.debian.net/hidden/92cb01ba/)
- sdcard structure by `tree`: [link](https://paste.debian.net/hidden/8426a629/)

What I find odd is that booting to emuMMC through Hekate SOMETIMES works (usually on the first clean try), but other times it just repeats the described behavior (especially after a reboot). 
Trying to boot to emuMMC by launching fusee-primary through Hekate also does not work at all.

Actions which did not help: reformatting sdcard, remaking emuMMC partition, checking the archive bit on sdcard files (Nintendo folders excluded), same setup on an entirely different sdcard.
Making the entire setup again on a new sdcard seemed to have fixed it at first but after a few reboots the same problem occurs (no changes to sdcard inbetween reboots)

m4xw, 2020-04-15:
When it does reboot 2 rcm and you inject fusee afterwards, it should display a abort error code (if emummc encounters a error).
Can you confirm & give me the output?
Also what SD card do you have?

veesonic, 2020-04-17:
I have also been experiencing similar behavior. I use Kosmos and reported an issue on their github. I have a screenshot of what the log shows when attempting to boot into emummc, I also get a black screen.

https://github.com/AtlasNX/Kosmos/issues/274

(It looks like there were a couple of duplicate issues. Sorry if posting this is against rules, this comment can be deleted if needed.)

brnor, 2020-04-17:
Sorry, been busy with work.

> 
> 
> When it does reboot 2 rcm and you inject fusee afterwards, it should display a abort error code (if emummc encounters a error).
> Can you confirm & give me the output?

Can't confirm. Usually after it shuts-down (or reboots to RCM, maybe that's due to autorcm) the next boot continues normally, and then the behavior explained repeats after the following reboot.

>
>
>Also what SD card do you have?

Some UHS class 3, 128gb, I think it's toshiba (or samsung)

So after reporting this I've wiped the data partition and installed atmosphere 0.10.4 again with hekate and remade the emuMMC partition. I've also stopped using the power menu reboot option and instead started using the reboot_to_payload.nro homebrew to boot back to hekate. So far it seems to work without issues after several reboots, but I haven't been able to find the time to do some extensive testing

SciresM, 2020-04-20:
Closing because it was resolved by re-creating emummc partition. (reboot to payload and reboot via power menu use identical smc mechanism btw)

closed 2020-04-20 by SciresM